stack-of-tasks/pinocchio #1233: CppADCogen couldn't found in recent devel branch

cmastalli, 2020-06-08:
Hi guys,

I want to report this issue that I only encounter in the devel branch, just in case you're not aware of it.
This is the current error message:
```
CMake Error at cmake/package-config.cmake:82 (find_package):
  By not providing "Findcppadcg.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "cppadcg", but
  CMake did not find one.

  Could not find a package configuration file provided by "cppadcg"
  (requested version 2.4.1) with any of the following names:

    cppadcgConfig.cmake
    cppadcg-config.cmake

  Add the installation prefix of "cppadcg" to CMAKE_PREFIX_PATH or set
  "cppadcg_DIR" to a directory containing one of the above files.  If
  "cppadcg" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:115 (ADD_PROJECT_DEPENDENCY)
```

jcarpent, 2020-06-08:
This is expected because CppADCG does not come with cmake module. Could you fix it? Thanks in advance.

nim65s, 2020-06-08:
For reference: https://github.com/loco-3d/solo-collisions/tree/master/cmake-modules

nim65s, 2020-06-08:
@jcarpent : should I put that in the jrl-cmakemodule ?

jcarpent, 2020-06-09:
I would say yes. Thanks a lot @nim65s.

cmastalli, 2020-06-09:
thank @nim65s for handling this task :)

closed 2020-06-10 by jcarpent