stack-of-tasks/pinocchio #1237: update of pinocchio package

mfocchi, 2020-06-11:
Hi 

I updated my pinocchio library (version 2.4.3) to the last one (version 2.4.5)
by running 
`sudo apt-get install robotpkg-py27-pinocchio`

for some reason this didn't install also **robotpkg-pinocchio** that seems 
to update the libraries and the sym links in /opt/openrobots/lib
therefore, when you try to import pinocchio got stuck because libpinocchio_pywrap it cannot find the right version of pinocchio cause libpinocchio.so  still points to libpinocchio.so.2.4.3

jcarpent, 2020-06-11:
Hi @mfocchi,

On which version of Linux are you working on now?
In the meantime, you can maybe force the installation of pinocchio on the 2.4.5 version via:
```
sudo apt install robotpkg-pinocchio=2.4.5
```
@nim65s Do you know what happens here?
Best,

Justin

nim65s, 2020-06-11:
@mfocchi : you can fix this with a `sudo apt upgrade`.
But yes, there is an issue in the packaging:
```bash
╰─>$ apt depends robotpkg-py27-pinocchio
[…]
 |Dépend: robotpkg-pinocchio (= 2.4.4)
 |Dépend: robotpkg-pinocchio (= 2.4.3)
 |Dépend: robotpkg-pinocchio (= 2.3.1)
  Dépend: robotpkg-pinocchio (= 2.4.5)
```

I'll fix this.

mfocchi, 2020-06-11:
> Hi @mfocchi,
> 
> On which version of Linux are you working on now?
> In the meantime, you can maybe force the installation of pinocchio on the 2.4.5 version via:
> 
> ```
> sudo apt install robotpkg-pinocchio=2.4.5
> ```
> 
> @nim65s Do you know what happens here?
> Best,
> 
> Justin

I am on ubuntu 16.04, and that is what I did to force it without any luck. I am glad the issue was sorted out!

best,
Michele

jcarpent, 2020-06-18:
Seems to be solved now. Thanks @mfocchi for raising this issue and to @nim65s for the fix.

closed 2020-06-18 by jcarpent