stack-of-tasks/pinocchio #1239: C++ visualization

jmirabel, 2020-06-12:
This is just an announce that may interest people using Pinocchio in C++.

I created https://github.com/jmirabel/pinocchio-gepetto-viewer/ that enables visualization of pinocchio Model in gepetto-viewer directly in C++.

At the moment, it depends on the devel branch of gepetto-viewer (not released yet).

jcarpent, 2020-06-12:
Maybe, you can refer to it in the documentation and in the Readme, by adding a list of the supported viewer.

cmastalli, 2020-06-12:
It would be great (for public visibility) if you, at some point, transferred the repository to `stack-of-tasks` organization.

cmastalli, 2020-06-12:
Btw, I know I asked this before, but I would like to make again the same question. Is there an improvement in computation performance?

jmirabel, 2020-06-12:
> Btw, I know I asked this before, but I would like to make again the same question. Is there an improvement in computation performance?

This wasn't my target. There are places where the code could be optimized.
If that worries you, we can put the viewer::display code in a separate thread.

jcarpent, 2020-08-17:
This information now appears on the Readme.md.

closed 2020-08-17 by jcarpent

jmirabel, 2020-08-18:
Thank you for taking care of this. While reading through the README, I bumped into the lines

> NEWS: We are hiring! We are looking for a talented C++/Python software developer to enhance and promote Pinocchio inside the robotics community and beyond. Please contact @jcarpent for further details.

Next time this file is updated, we may put a date on this so that people know it is recent.

jmirabel, 2021-03-05:
A follow-up: https://github.com/stack-of-tasks/pinocchio-gepetto-viewer/pull/2