stack-of-tasks/pinocchio #1250: Pinocchio Mac Import Error: (libboost_python38-mt.dylib)

AtharvaSune, 2020-06-19:
```
Python 3.7.5 (v3.7.5:5c02a39a0b, Oct 14 2019, 18:49:57) 
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/opt/pinocchio/lib/python3.8/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .libpinocchio_pywrap import *
ImportError: dlopen(/usr/local/opt/pinocchio/lib/python3.8/site-packages/pinocchio/libpinocchio_pywrap.so, 2): Symbol not found: __Py_tracemalloc_config
  Referenced from: /usr/local/opt/boost-python3/lib/libboost_python38-mt.dylib
  Expected in: flat namespace
 in /usr/local/opt/boost-python3/lib/libboost_python38-mt.dylib
```
Using MacOS Catalina, python 3.7.5

jcarpent, 2020-06-19:
You should use python3.8, on OS X, we made the choice of python3.8.

jcarpent, 2020-06-19:
If you want, you can use Conda to get the 3.7 version of the lib.

jcarpent, 2020-06-22:
@AtharvaSune I think this issue is solved. Please reopen it if needed.

closed 2020-06-22 by jcarpent