stack-of-tasks/pinocchio #1251: How to set joint limits in inverse kinematics

zhangOSK, 2020-06-21:
Hi there, I am using the Jacobian-based iterative IK in Pinocchio.  
I want to ask if pinocchio can consider the joint limits when calculating IK?

jcarpent, 2020-06-22:
Of course, you can, you just need to handle them through a QP-based approach and to use the `difference` function to compute the distance between two configurations.

zhangOSK, 2020-06-23:
Thank you for the reply!
1. Could you please give me more description about using QP-based approach? Like some sample code or the source code related? I checked the function computeJointJacobian but didn't find anything about qp.
2. I want to make sure the difference function, do you mean the following function?
https://github.com/stack-of-tasks/pinocchio/blob/a179f5619a24000d71d748243b5068605cce4453/src/algorithm/joint-configuration.hpp#L121

olivier-stasse, 2020-06-23:
Dear @zhangOSK,
pinocchio is dedicated to rigid body dynamics. They are some simple mechanisms to compute IK, but the two main libraries build on top on pinocchio to perform control are either [TSID](https://github.com/stack-of-tasks/tsid/) for torque or acceleration QP based, or [sot-core](https://github.com/stack-of-tasks/sot-core) for IK. For the latter, you can install the binaries to make Talos walk forward in real-time by looking at [talos_integration_tests](https://github.com/stack-of-tasks/talos_integration_tests). Note that with the latest release, the robot is still unstable while turning.

zhangOSK, 2020-06-23:
Dear Olivier, thank you very much for the explanation! Now I know where to check!

closed 2020-06-23 by zhangOSK