stack-of-tasks/pinocchio #1252: computeCentroidalMomentum()

rshum19, 2020-06-29:
Hello, 

I'm trying to get familiarized with the library to use it as a part of a trajectory optimization implementation based on centroidal dynamics. Looking through the doxygen documentation I came across the function `computeCentroidalMomentum()`.  

I want to double-check that this function computes the Centroidal Momentum of a Rigid Body Tree as described in 

- *"Centroidal dynamics of a Humanoid Robot" - Orin, Goswami and Lee 2013*

If this is not the case, what is this function exactly computing?

jcarpent, 2020-06-29:
Yes. This exactly what it does. It compute the total angular momentum quantity expressed in the frame centered at the center of mass and with axes aligned to the world frame.
To compute also compute the time derivative, you can call computeCentroidalMomentumTimeVariation

jcarpent, 2020-06-30:
I will close this as it seems to be solved. Feel free to reopen it if needed.

closed 2020-06-30 by jcarpent

rshum19, 2020-06-30:
Thank you for the clarification and prompt response. these functions are great!! This has solved my query.

jcarpent, 2020-06-30:
Cool. Glad to see Pinocchio fullfiled your needs.

rshum19, 2020-10-25:
Hi @jcarpent 

Will `computeCentroidalMomentum()` and  `computeCentroidalMomentumTimeVariation()` still be valid for fixed based systems (e.g. inverted pendulum) or it has to be a free floating base robot?

jcarpent, 2020-10-26:
The centroidal dynamics refers to under-actuated systems equipped with a floating or planar joint.
But, you can still use these quantities in the context of other robots, you just need to be aware of the computations won't account for the base, because it is not moving (it is fixed).

rshum19, 2020-10-26:
First I want to thank for the prompt response! 

I’m somewhat unclear about the  statement 

> you just need to be aware of the computations won't account for the base,

What is the the model is an under actuated system like a inverted pendulum on a cart.  Where we model motion of the cart with a prismatic joint and the motion of the pendulum with a revolute joint. Only the prismatic joint is actuated. 

Assuming the prismatic joint is in along the X axis. Then if we compute the centroidal momentum h_g, will the rows corresponding to the Y axis direction motion be zero because there is no motion in this direction? Is this correct?

cmastalli, 2020-10-26:
> What is the the model is an under actuated system like a inverted pendulum on a cart. Where we model motion of the cart with a prismatic joint and the motion of the pendulum with a revolute joint. Only the prismatic joint is actuated.

This is not a fixed based system, it is still a floating base but with less degrees of freedom.

jcarpent, 2020-10-26:
In my previous comments, I was referring to robotic arm, which has a base which as a certain inertia. But the inertia of the base won't be taken into account when computing the centroidal dynamics, because it does not move.

On the other side, if you put this robotic arm on a mobile basis (aka planar joint), then the inertia of the base of the robotic arm will contribute to the centroidal dynamics of the whole system.

rshum19, 2020-10-26:
> This is not a fixed based system, it is still a floating base but with less degrees of freedom.

@cmastalli I agree, I should had been more explicit to say that I was abusing naming and referring to a "Fixed base" to a model I did not explicitly add a `pinocchio::JointModelFreeFlyer()` joint when building the model from URDF.

@jcarpent That makes sense. Going off my comment above, is it equivalent if I define the mobile basis joints in the URDF than if I did something like this 

```
pinocchio::Model model;
pinocchio::urdf::buildModel(urdf_filename, pinocchio::JointModelPlanar(), model);
```

jcarpent, 2020-10-26:
Yes, you did well.

rshum19, 2020-10-26:
@jcarpent ok great! As always thank you for your prompt and helpful response.

edward9503, 2022-02-26:
> Yes. This exactly what it does. It computes the total angular momentum quantity expressed in the frame centered at the center of mass and with axes aligned to the world frame. To compute also the time derivative, you can call computeCentroidalMomentumTimeVariation

Hi @jcarpent , just suddenly one short question: You have mentioned that computeCentroidalMomentum computes the total angular momentum quantity expressed in the frame centered at the center of mass and with axes aligned to the world frame. Is it also the same for the linear momentum?

Thanks in advance.

jcarpent, 2022-02-26:
Yes, sure.