stack-of-tasks/pinocchio #1254: uninstalled packages during installing pinnochio using robotpkg

ahmadgazar, 2020-07-13:
Hi,
I am using ubuntu 18.04, and I tried installing pinocchio using robot-pkg with python 3.6. I then tried to import pinocchio, but turns out that libhpp-fcl and octomap were not installed during the pinocchio installation. Problem was solved by manually reinstalling the two packages as:

$ sudo apt install --reinstall robotpkg-hpp-fcl
$ sudo apt install --reinstall robotpkg-octomap

I thought this might be helpful for someone who had a similar problem.

jcarpent, 2020-07-15:
They should be installed along with pinocchio. @nim65s Could you provide some feedback on this issue?

jcarpent, 2020-08-14:
@nim65s ping.

nim65s, 2020-08-14:
I don't unsterdand this issue, as this is working for me in a clean container:
```bash
root@5e832ab87bff:~/robotpkg# apt depends robotpkg-py36-pinocchio
robotpkg-py36-pinocchio
 |Depends: robotpkg-py36-hpp-fcl (= 1.4.5)
 |Depends: robotpkg-py36-hpp-fcl (= 1.4.4r1)
 |Depends: robotpkg-py36-hpp-fcl (= 1.2.1)
 |Depends: robotpkg-py36-hpp-fcl+doc (= 1.2.0)
 |Depends: robotpkg-py36-hpp-fcl (= 1.4.4)
 |Depends: robotpkg-py36-hpp-fcl (= 1.4.2)
 |Depends: robotpkg-py36-hpp-fcl (= 1.2.0)
 |Depends: robotpkg-py36-hpp-fcl (= 1.3.0)
  Depends: robotpkg-py36-hpp-fcl (= 1.2.2)
 |Depends: robotpkg-py36-eigenpy (= 2.3.0)
 |Depends: robotpkg-py36-eigenpy (= 2.4.0)
 |Depends: robotpkg-py36-eigenpy (= 2.2.0r1)
 |Depends: robotpkg-py36-eigenpy (= 2.4.3)
  Depends: robotpkg-py36-eigenpy (= 2.2.0)
  Depends: robotpkg-pinocchio (= 2.4.7)
 |Depends: robotpkg-octomap (= 1.6.1)
  Depends: robotpkg-octomap (= 1.9.0)
 |Depends: robotpkg-hpp-fcl+doc (= 1.4.2)
 |Depends: robotpkg-hpp-fcl (= 1.4.2)
 |Depends: robotpkg-hpp-fcl (= 1.4.5)
 |Depends: robotpkg-hpp-fcl+doc (= 1.4.4)
 |Depends: robotpkg-hpp-fcl+doc (= 1.4.5)
  Depends: robotpkg-hpp-fcl (= 1.4.4)
  Depends: libboost-python1.65-dev (>= 1.34.1)
  Depends: libstdc++6
  Depends: python3-numpy (>= 1:1)
  Depends: python3.6-minimal (>= 3.6)
  Depends: python3.6-minimal (<< 3.7)
  Depends: libpython3.6-dev (>= 3.6)
  Depends: libpython3.6-dev (<< 3.7)
  Depends: libtinyxml-dev (>= 2)
  Depends: liburdfdom-dev (>= 0.2.8)
  Conflicts: robotpkg-py36-pinocchio
  Replaces: robotpkg-py36-pinocchio
```

jcarpent, 2020-08-14:
OK. I will close this issue then. Thanks for your feedback @nim65s.

closed 2020-08-14 by jcarpent