stack-of-tasks/pinocchio #1258: Include a function for updating the forward dynamics (updateForwardDynamics)

cmastalli, 2020-07-20:
This issue proposes a new signature for updating the forward dynamics. Currently, `forwardDynamics` runs internally `computeAllTerms`, however, there are cases where the user might want to decouple both things. It gives the freedom to update manually these terms.
Indeed, there is will improve the efficiency of some internal computations in Crocoddyl (see https://github.com/loco-3d/crocoddyl/blob/master/include/crocoddyl/multibody/actions/contact-fwddyn.hxx#L75). However, and generally speaking, you need to compute the Jacobians that requires to compute twice them.

proyan, 2020-07-20:
Hi Carlos,
I think it is already there.
https://github.com/stack-of-tasks/pinocchio/blob/master/src/algorithm/contact-dynamics.hpp#L86 ?

proyan, 2020-07-20:
In addition to this signature, there is a new API and algorithm that Justin might push in the coming weeks. So I would suggest that you don't spend a lot of time to optimize crocoddyl's contact dynamics, as that would also need to be updated :).

jcarpent, 2020-07-20:
I share the same opinion as @proyan. I'm entirely refactoring the contact dynamics API with new features, better robustness and faster than the current version. People just need to be a bit patient.

cmastalli, 2020-07-20:
> I think it is already there.
> https://github.com/stack-of-tasks/pinocchio/blob/master/src/algorithm/contact-dynamics.hpp#L86 ?

I see, thanks

> In addition to this signature, there is a new API and algorithm that Justin might push in the coming weeks. So I would suggest that you don't spend a lot of time to optimize crocoddyl's contact dynamics, as that would also need to be updated :).
> I share the same opinion as @proyan. I'm entirely refactoring the contact dynamics API with new features, better robustness and faster than the current version. People just need to be a bit patient.

I know you're working on it. But it might take additional time to have a working version in Crocoddyl. Turning off the flag it is a simple patch for the moment.

I will close this issue. Thanks for the answer!

closed 2020-07-20 by cmastalli