stack-of-tasks/pinocchio #1259: Linearized dynamics

rshum19, 2020-07-22:
Hello, 

I'm wondering if Pinocchio has methods to output the linearized dynamics of the system. I see CppAd and Casadi are dependencies of the library, thus wondering if this has been used to obtain either analytical or numerical linearized dynamics. 

Thank you,
Roberto

jcarpent, 2020-07-22:
Dear @rshum19,

Indeed, Pinocchio provides the analytical derivatives of standard algorithms for computing forward and inverse dynamics, as well as the derivatives of the kinematic quantities.
There is some tiny example in the example folder.

jcarpent, 2020-07-22:
In addition, if you need more derivatives, I may take some time to implement them.

rshum19, 2020-07-22:
Hi @jcarpent 

Great I think this is exactly what I was looking for. I’ll look more into this derivative. 

Thank you!

closed 2020-07-22 by rshum19