stack-of-tasks/pinocchio #1261: Comilation error with the overview example

CWEzio, 2020-07-27:
Hi, I am new to pinocchio and have tried the overview examples. However, I encounter compilation errors with both examples. The error seems to be related with Eigen. I install pinocchio follow the instruction here https://stack-of-tasks.github.io/pinocchio/download.html. The eigen version I used is 3.3.7. The system is Ubuntu 18.04.

Here is the detailed info
```
>> g++ -I /usr/include/eigen3 -I /opt/openrobots/include/ overview_simple.cpp -o overview-simple 

In file included from /usr/include/eigen3/unsupported/Eigen/CXX11/Tensor:105:0,
                 from /opt/openrobots/include/pinocchio/fwd.hpp:31,
                 from /opt/openrobots/include/pinocchio/spatial/fwd.hpp:9,
                 from /opt/openrobots/include/pinocchio/multibody/model.hpp:9,
                 from /opt/openrobots/include/pinocchio/parsers/sample-models.hpp:9,
                 from overview_simple.cpp:1:
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorEvaluator.h: In instantiation of ‘struct Eigen::TensorEvaluator<const Eigen::Tensor<double, 3, 0, long int>, Eigen::DefaultDevice>’:
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorEvaluator.h:232:8:   required from ‘struct Eigen::TensorEvaluator<const Eigen::TensorCwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Tensor<double, 3, 0, long int> >, Eigen::DefaultDevice>’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorAssign.h:96:70:   required from ‘struct Eigen::TensorEvaluator<const Eigen::TensorAssignOp<Eigen::Tensor<double, 3, 0, long int>, const Eigen::TensorCwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Tensor<double, 3, 0, long int> > >, Eigen::DefaultDevice>’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/Tensor.h:416:14:   required from ‘Eigen::Tensor<Scalar_, NumIndices_, Options_, IndexType>& Eigen::Tensor<Scalar_, NumIndices_, Options_, IndexType>::operator=(const OtherDerived&) [with OtherDerived = Eigen::TensorCwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Tensor<double, 3, 0, long int> >; Scalar_ = double; int NumIndices_ = 3; int Options_ = 0; IndexType_ = long int]’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBase.h:845:24:   required from ‘Derived& Eigen::TensorBase<Derived, AccessLevel>::setConstant(const Scalar&) [with Derived = Eigen::Tensor<double, 3, 0, long int>; int AccessLevel = 1; Eigen::TensorBase<Derived, AccessLevel>::Scalar = double]’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBase.h:841:25:   required from ‘Derived& Eigen::TensorBase<Derived, AccessLevel>::setZero() [with Derived = Eigen::Tensor<double, 3, 0, long int>; int AccessLevel = 1]’
/opt/openrobots/include/pinocchio/multibody/data.hxx:133:5:   required from ‘pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>::DataTpl(const Model&) [with _Scalar = double; int _Options = 0; JointCollectionTpl = pinocchio::JointCollectionDefaultTpl; pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>::Model = pinocchio::ModelTpl<double>]’
overview_simple.cpp:8:29:   required from here
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorEvaluator.h:162:71: warning: ignoring attributes on template argument ‘Eigen::PacketType<double, Eigen::DefaultDevice>::type {aka __vector(2) double}’ [-Wignored-attributes]
     PacketAccess = (internal::unpacket_traits<PacketReturnType>::size > 1),
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
In file included from /usr/include/eigen3/unsupported/Eigen/CXX11/Tensor:105:0,
                 from /opt/openrobots/include/pinocchio/fwd.hpp:31,
                 from /opt/openrobots/include/pinocchio/spatial/fwd.hpp:9,
                 from /opt/openrobots/include/pinocchio/multibody/model.hpp:9,
                 from /opt/openrobots/include/pinocchio/parsers/sample-models.hpp:9,
                 from overview_simple.cpp:1:
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorEvaluator.h: In instantiation of ‘struct Eigen::TensorEvaluator<Eigen::Tensor<double, 3, 0, long int>, Eigen::DefaultDevice>’:
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorAssign.h:100:65:   required from ‘struct Eigen::TensorEvaluator<const Eigen::TensorAssignOp<Eigen::Tensor<double, 3, 0, long int>, const Eigen::TensorCwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Tensor<double, 3, 0, long int> > >, Eigen::DefaultDevice>’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/Tensor.h:416:14:   required from ‘Eigen::Tensor<Scalar_, NumIndices_, Options_, IndexType>& Eigen::Tensor<Scalar_, NumIndices_, Options_, IndexType>::operator=(const OtherDerived&) [with OtherDerived = Eigen::TensorCwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Tensor<double, 3, 0, long int> >; Scalar_ = double; int NumIndices_ = 3; int Options_ = 0; IndexType_ = long int]’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBase.h:845:24:   required from ‘Derived& Eigen::TensorBase<Derived, AccessLevel>::setConstant(const Scalar&) [with Derived = Eigen::Tensor<double, 3, 0, long int>; int AccessLevel = 1; Eigen::TensorBase<Derived, AccessLevel>::Scalar = double]’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBase.h:841:25:   required from ‘Derived& Eigen::TensorBase<Derived, AccessLevel>::setZero() [with Derived = Eigen::Tensor<double, 3, 0, long int>; int AccessLevel = 1]’
/opt/openrobots/include/pinocchio/multibody/data.hxx:133:5:   required from ‘pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>::DataTpl(const Model&) [with _Scalar = double; int _Options = 0; JointCollectionTpl = pinocchio::JointCollectionDefaultTpl; pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>::Model = pinocchio::ModelTpl<double>]’
overview_simple.cpp:8:29:   required from here
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorEvaluator.h:42:71: warning: ignoring attributes on template argument ‘Eigen::PacketType<double, Eigen::DefaultDevice>::type {aka __vector(2) double}’ [-Wignored-attributes]
     PacketAccess = (internal::unpacket_traits<PacketReturnType>::size > 1),
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
In file included from /usr/include/eigen3/unsupported/Eigen/CXX11/Tensor:139:0,
                 from /opt/openrobots/include/pinocchio/fwd.hpp:31,
                 from /opt/openrobots/include/pinocchio/spatial/fwd.hpp:9,
                 from /opt/openrobots/include/pinocchio/multibody/model.hpp:9,
                 from /opt/openrobots/include/pinocchio/parsers/sample-models.hpp:9,
                 from overview_simple.cpp:1:
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorExecutor.h: In instantiation of ‘static void Eigen::internal::TensorExecutor<Expression, Eigen::DefaultDevice, true>::run(const Expression&, const Eigen::DefaultDevice&) [with Expression = const Eigen::TensorAssignOp<Eigen::Tensor<double, 3, 0, long int>, const Eigen::TensorCwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Tensor<double, 3, 0, long int> > >]’:
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/Tensor.h:417:65:   required from ‘Eigen::Tensor<Scalar_, NumIndices_, Options_, IndexType>& Eigen::Tensor<Scalar_, NumIndices_, Options_, IndexType>::operator=(const OtherDerived&) [with OtherDerived = Eigen::TensorCwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Tensor<double, 3, 0, long int> >; Scalar_ = double; int NumIndices_ = 3; int Options_ = 0; IndexType_ = long int]’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBase.h:845:24:   required from ‘Derived& Eigen::TensorBase<Derived, AccessLevel>::setConstant(const Scalar&) [with Derived = Eigen::Tensor<double, 3, 0, long int>; int AccessLevel = 1; Eigen::TensorBase<Derived, AccessLevel>::Scalar = double]’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBase.h:841:25:   required from ‘Derived& Eigen::TensorBase<Derived, AccessLevel>::setZero() [with Derived = Eigen::Tensor<double, 3, 0, long int>; int AccessLevel = 1]’
/opt/openrobots/include/pinocchio/multibody/data.hxx:133:5:   required from ‘pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>::DataTpl(const Model&) [with _Scalar = double; int _Options = 0; JointCollectionTpl = pinocchio::JointCollectionDefaultTpl; pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>::Model = pinocchio::ModelTpl<double>]’
[... cut ...]
example.cpp:(.text._ZN9pinocchio4urdf10buildModelIdLi0ENS_25JointCollectionDefaultTplEEERNS_8ModelTplIT_XT0_ET1_EERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_b[_ZN9pinocchio4urdf10buildModelIdLi0ENS_25JointCollectionDefaultTplEEERNS_8ModelTplIT_XT0_ET1_EERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_b]+0x57): undefined reference to `pinocchio::urdf::details::parseRootTree(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::urdf::details::UrdfVisitorBaseTpl<double, 0>&)'
collect2: error: ld returned 1 exit status
```

jmirabel, 2020-07-27:
This is a linking problem. You should link to Pinocchio library too.

Why don't you use CMake ? It would make this transparent for you.

jcarpent, 2020-07-27:
You may also use the help of pkg-config for solving your issue:
```
g++ $(pkg-config --cflags pinocchio) $(pkg-config --libs pinocchio) overview_simple.cpp -o overview-simple 
```

CWEzio, 2020-07-27:
Thanks a lot for both of your help. Using cmake solves my problem. The reason that I didn't use cmake at first is that in the documentation (https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/index.html), they use g++ to compile the example and I simply follows their procedures.

Here is the cmakelists.txt file I write, someone new to pinocchio and c++ may find it helpful.
```cmake
CMAKE_MINIMUM_REQUIRED(VERSION 3.5)
project(overview_simple)

find_package(Eigen3 REQUIRED NO_MODULE)
list(APPEND CMAKE_PREFIX_PATH "/opt/openrobots/")
find_package(pinocchio REQUIRED)

add_executable(${PROJECT_NAME} overview_simple.cpp)
target_link_libraries(${PROJECT_NAME} PRIVATE Eigen3::Eigen pinocchio::pinocchio)

add_executable(overview_complex example.cpp)
target_link_libraries(overview_complex PRIVATE Eigen3::Eigen pinocchio::pinocchio)
```

CWEzio, 2020-07-27:
> You may also use the help of pkg-config for solving your issue:
> 
> ```
> g++ $(pkg-config --cflags pinocchio) $(pkg-config --libs pinocchio) overview_simple.cpp -o overview-simple 
> ```

Sadly, this also fails in my computer. It seems that boost_system lib is not linked correctly, though my LD_LIBRARY_PATH contains /usr/lib/x86_64-linux-gnu where the libboost_system.so is. Using cmake directly does not have this issue. It may because cmake finds this lib automatically. But I am still curious about why using g++  fails.....
```
In file included from /usr/include/eigen3/unsupported/Eigen/CXX11/Tensor:105:0,
                 from /opt/openrobots/include/pinocchio/fwd.hpp:31,
                 from /opt/openrobots/include/pinocchio/spatial/fwd.hpp:9,
                 from /opt/openrobots/include/pinocchio/multibody/model.hpp:9,
                 from /opt/openrobots/include/pinocchio/parsers/sample-models.hpp:9,
                 from overview_simple.cpp:1:
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorEvaluator.h: In instantiation of ‘struct Eigen::TensorEvaluator<const Eigen::Tensor<double, 3, 0, long int>, Eigen::DefaultDevice>’:
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorEvaluator.h:232:8:   required from ‘struct Eigen::TensorEvaluator<const Eigen::TensorCwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Tensor<double, 3, 0, long int> >, Eigen::DefaultDevice>’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorAssign.h:96:70:   required from ‘struct Eigen::TensorEvaluator<const Eigen::TensorAssignOp<Eigen::Tensor<double, 3, 0, long int>, const Eigen::TensorCwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Tensor<double, 3, 0, long int> > >, Eigen::DefaultDevice>’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/Tensor.h:416:14:   required from ‘Eigen::Tensor<Scalar_, NumIndices_, Options_, IndexType>& Eigen::Tensor<Scalar_, NumIndices_, Options_, IndexType>::operator=(const OtherDerived&) [with OtherDerived = Eigen::TensorCwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Tensor<double, 3, 0, long int> >; Scalar_ = double; int NumIndices_ = 3; int Options_ = 0; IndexType_ = long int]’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBase.h:845:24:   required from ‘Derived& Eigen::TensorBase<Derived, AccessLevel>::setConstant(const Scalar&) [with Derived = Eigen::Tensor<double, 3, 0, long int>; int AccessLevel = 1; Eigen::TensorBase<Derived, AccessLevel>::Scalar = double]’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBase.h:841:25:   required from ‘Derived& Eigen::TensorBase<Derived, AccessLevel>::setZero() [with Derived = Eigen::Tensor<double, 3, 0, long int>; int AccessLevel = 1]’
/opt/openrobots/include/pinocchio/multibody/data.hxx:133:5:   required from ‘pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>::DataTpl(const Model&) [with _Scalar = double; int _Options = 0; JointCollectionTpl = pinocchio::JointCollectionDefaultTpl; pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>::Model = pinocchio::ModelTpl<double>]’
overview_simple.cpp:8:29:   required from here
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorEvaluator.h:162:71: warning: ignoring attributes on template argument ‘Eigen::PacketType<double, Eigen::DefaultDevice>::type {aka __vector(2) double}’ [-Wignored-attributes]
     PacketAccess = (internal::unpacket_traits<PacketReturnType>::size > 1),
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
In file included from /usr/include/eigen3/unsupported/Eigen/CXX11/Tensor:105:0,
                 from /opt/openrobots/include/pinocchio/fwd.hpp:31,
                 from /opt/openrobots/include/pinocchio/spatial/fwd.hpp:9,
                 from /opt/openrobots/include/pinocchio/multibody/model.hpp:9,
                 from /opt/openrobots/include/pinocchio/parsers/sample-models.hpp:9,
                 from overview_simple.cpp:1:
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorEvaluator.h: In instantiation of ‘struct Eigen::TensorEvaluator<Eigen::Tensor<double, 3, 0, long int>, Eigen::DefaultDevice>’:
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorAssign.h:100:65:   required from ‘struct Eigen::TensorEvaluator<const Eigen::TensorAssignOp<Eigen::Tensor<double, 3, 0, long int>, const Eigen::TensorCwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Tensor<double, 3, 0, long int> > >, Eigen::DefaultDevice>’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/Tensor.h:416:14:   required from ‘Eigen::Tensor<Scalar_, NumIndices_, Options_, IndexType>& Eigen::Tensor<Scalar_, NumIndices_, Options_, IndexType>::operator=(const OtherDerived&) [with OtherDerived = Eigen::TensorCwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, const Eigen::Tensor<double, 3, 0, long int> >; Scalar_ = double; int NumIndices_ = 3; int Options_ = 0; IndexType_ = long int]’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBase.h:845:24:   required from ‘Derived& Eigen::TensorBase<Derived, AccessLevel>::setConstant(const Scalar&) [with Derived = Eigen::Tensor<double, 3, 0, long int>; int AccessLevel = 1; Eigen::TensorBase<Derived, AccessLevel>::Scalar = double]’
/usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBase.h:841:25:   required from ‘Derived& Eigen::TensorBase<Derived, AccessLevel>::setZero() [with Derived = Eigen::Tensor<double, 3, 0, long int>; int  [... cut ...]
overview_simple.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0xf3): undefined reference to `boost::system::generic_category()'
overview_simple.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x1d2): undefined reference to `boost::system::generic_category()'
collect2: error: ld returned 1 exit status
```

jcarpent, 2020-07-27:
Can you output the result of:
```
pkg-config --libs pinocchio
```

CWEzio, 2020-07-27:
```
>> echo $(pkg-config --libs pinocchio)  
-L/opt/openrobots/lib -Wl,-rpath,/opt/openrobots/lib -lpinocchio -Wl,-rpath,/usr/lib/x86_64-linux-gnu -lboost_filesystem -lboost_serialization -lboost_system -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -lconsole_bridge -Wl,-rpath,/opt/openrobots/lib -lhpp-fcl -Wl,-rpath,/usr/lib/x86_64-linux-gnu -lboost_thread -lboost_date_time -lboost_system -loctomap -loctomath

```

jcarpent, 2020-07-27:
@nim65s Could you try it on your side?

jmirabel, 2020-07-28:
FYI CMake does it in two steps:
```
[ 50%] Building CXX object CMakeFiles/overview-simple.dir/overview-simple.cpp.o
/usr/bin/g++  -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DHPP_FCL_HAVE_OCTOMAP -DOCTOMAP_MAJOR_VERSION=1 -DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=0 -DPINOCCHIO_WITH_HPP_FCL -DPINOCCHIO_WITH_URDFDOM -isystem /usr/include/eigen3 -isystem /home/jmirabel/devel/hpp/install/include -isystem /opt/ros/melodic/include   -o CMakeFiles/overview-simple.dir/overview-simple.cpp.o -c /home/jmirabel/devel/hpp/src/test/pinocchio/overview-simple.cpp
[100%] Linking CXX executable overview-simple
/usr/bin/g++     CMakeFiles/overview-simple.dir/overview-simple.cpp.o  -o overview-simple -Wl,-rpath,/home/jmirabel/devel/hpp/install/lib:/opt/ros/melodic/lib /home/jmirabel/devel/hpp/install/lib/libpinocchio.so.2.4.6 /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_serialization.so /usr/lib/x86_64-linux-gnu/liburdfdom_sensor.so /usr/lib/x86_64-linux-gnu/liburdfdom_model_state.so /usr/lib/x86_64-linux-gnu/liburdfdom_model.so /usr/lib/x86_64-linux-gnu/liburdfdom_world.so /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 /home/jmirabel/devel/hpp/install/lib/libhpp-fcl.so /opt/ros/melodic/lib/liboctomap.so /opt/ros/melodic/lib/liboctomath.so 
```

If you want to do it in one step, you must put the -l arguments at the end:
`g++ $(pkg-config --cflags pinocchio) ../overview-simple.cpp -o overview-simple $(pkg-config --libs pinocchio)`

nim65s, 2020-07-28:
I can reproduce @CWEzio output.

jcarpent, 2020-07-28:
Thanks @jmirabel for the feedback. This issue seems to be solved then.

closed 2020-07-28 by jcarpent