stack-of-tasks/pinocchio #1263: model with floating joint in urdf cannot be parsed

Wenchun6, 2020-07-30:
I want to model a floating base articulated body robot, such as humanoid or quadruped. Refer to the URDF official document, I found a joint label "floating" to discribe a floating base. But it shown a error of "The XML stream does not contain a valid URDF model" after I excute the program. While it can success run if the "floating" joint become "fix", but in this way, it seems not a dynamics system I wanted. So my question is that, how can I model a floating base robot in pinocchio, and I note that none of robot described with URDF has no such "floating" joint at all, why?

jcarpent, 2020-07-30:
@Wenchun6 could you please share with us your URDF file to check the issue you have encountered? In the meantime, you might see at the examples folder located at the root of the repository. There is a file called overview-URDF.py in which you will find the related example.

Wenchun6, 2020-07-30:
@jcarpent actually I have no problem to load my URDF model. So let me make it clear, I have a quadrped urdf model with 12 actuated revolute joints, by running the overview-URDF.cpp example as follow
[image]
The output is shown below,
[image]
It shows that my model is just a  fixed base model with 12 actuated joint, rather than a floating base model with nq = 6+12 and nv = 12, which the 6 corresponding to the degree of freedom in space of the floating base. I wanna ask for how can I model such floating base system in pinocchio, shoulde I need to add 3 prismatic and 3 revolute joints in my urdf or there is a specific function like FloatingBaseModel(model) or some thing others to model a floating base system?.

jcarpent, 2020-07-30:
You can just provide a third argument to the function buildModel like pinocchio::JointModelFreeFlye() to mention the fact that the root joint is a free floating joint.

jmirabel, 2020-07-30:
The problem comes from your side. With `test.urdf' being
```xml
<robot name="foo">
  <link name="link1"/>
  <link name="link2"/>
  <joint name="joint" type="floating">
    <parent link="link1"/>
    <child link="link2"/>
  </joint>
</robot>
```
then
```python
import pinocchio
model = pinocchio.buildModelFromUrdf("./test.urdf")
print(model)
```
outputs
```
Nb joints = 2 (nq=7,nv=6)
  Joint 0 universe: parent=0
  Joint 1 joint: parent=0
```
```

Wenchun6, 2020-07-30:
@jmirabel well,I still get the error if I set the root joint as a 'floating' joint in my urdf file. The error show that it is not valid urdf file

jmirabel, 2020-07-30:
The issue seems to be your URDF. You can check it with command `check_urdf`.

I close this since the issue does not seem to come from Pinocchio.

closed 2020-07-30 by jmirabel

Wenchun6, 2020-07-30:
@jcarpent That's great, thank you.So the first 7 parameters of 'q' mean the (x,y,z,quaternion), which is the position and orientation of the floating base? BTW,the quaternion in Pinocchio follows which conventions, (x,y,z,w) or (w,x,y,z)?