stack-of-tasks/pinocchio #1264: Issue related to usage of buildReducedModel in Crocoddyl

julesser, 2020-07-31:
Hi guys, I'm having issues in Crocoddyl that may or may not be related to Pinocchios `buildReducedModel `API. Consider the following two cases:
- Case 1: Use a full URDF version and apply **buildReducedModel** to lock the desired joints in a fixed position
- Case2: Use a URDF version, where the desired joints directly are set to `type="fixed"`

**Results**: For case1, the solver does not optimize anything (also no progress output) and the trajectories are static. For case 2, everything is working as expected. Apart from the model, nothing has been changed in both cases: 
[image]
[image]

**Debugging Attempts:** I've compared the resulting models via
```
print('reduced model: dim=' + str(len(rh5_robot.model.joints)))
for jn in rmodel.names:
    print(jn)
for jn in rmodel.joints:
    print(jn)
```
and both models appear to be identical from this perspective. See the following ouputs: 
[Case1BuildReducedModel.txt](https://github.com/stack-of-tasks/pinocchio/files/5006245/Case1BuildReducedModel.txt)
[Case2FixedJointsInURDF.txt](https://github.com/stack-of-tasks/pinocchio/files/5006247/Case2FixedJointsInURDF.txt)

However, if I see it correctly, the problem has to lie inside the model since this is the only changed parameter. Do you have any suggestions on how to debug this issue further?

jcarpent, 2020-07-31:
Maybe displaying the joint placements for a given joint configuration.

jmirabel, 2020-08-03:
I would try to make a test independent of Crocoddyl. Build the two models as you did and check that the output of the functions used by Crocoddyl are the same, on some elementary or random input.

julesser, 2020-08-03:
Thanks for your comments. I will check these tomorrow.

jcarpent, 2020-08-12:
@julesser Can you share your robot model?

julesser, 2020-08-14:
@jcarpent The problem is solved now, in fact it was an issue related to Crocoddyl. After building the latest devel branch from source, everything is working as expected. Thanks for your patience and kind support!

closed 2020-08-14 by julesser

jcarpent, 2020-08-14:
Nice to hear that your bug is solved.