stack-of-tasks/pinocchio #1269: Broken links in documentation

andreadelprete, 2020-08-06:
All the links to the code are broken in this page of the documentation:
https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_d-practical-exercises_7-learn.html

I've seen that the code is available [here](https://github.com/stack-of-tasks/pinocchio/tree/master/doc/d-practical-exercises/src), but anyway it'd be nice to fix these links.

jmirabel, 2020-08-06:
Since here, we can change *Learning to flight* to *Learn to fly* (*Learn* for consistency with the other practical exercises.

nim65s, 2020-08-07:
Those tutorials are currently not maintained, and they will be remplaced by a more recent version from Nicolas, as soon as I can spend some time to review and adapt them.

jcarpent, 2020-08-07:
Thanks @andreadelprete for raising this issue. I've updated doxygen to prevent it.
You can pull it and makde doc in your build directory to get the lattest version of the doc.

andreadelprete, 2020-08-07:
Ok, thanks! Feel free to close the issue if you want.

jcarpent, 2020-08-07:
Solved by #1270

closed 2020-08-07 by jcarpent

nim65s, 2020-08-08:
Fixed here: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/devel/doxygen-html/md_doc_d-practical-exercises_7-learn.html