stack-of-tasks/pinocchio #1277: About inverse rotation function

ddliu365, 2020-08-16:
Is it possible to implement the following the `inverse rotate` function?
```
m = rotate('x', pi / 3).dot(rotate('y', pi / 5)).dot(rotate('z', pi / 7))
pi/3, pi/5, pi/7 = inverseRotate(m)
```

jmirabel, 2020-08-17:
Check the rpy namespace (pinocchio.rpy of pinocchio::rpy).

jcarpent, 2020-08-17:
You can also use the Eigen functions for that. It is a direct binding of Eigen Geometry module in Python. 
See AngleAxis for that.

jcarpent, 2020-08-17:
There is also:
```python
import eigenpy
eigenpy.toEulerAngles(R)
```
with following doc:
```python
Docstring:
toEulerAngles( (object)mat (dim 3x3), (int)a0, (int)a1, (int)a2) -> object :
    It returns the Euler-angles of the rotation matrix mat using the convention defined by the triplet (a0,a1,a2).
```

ddliu365, 2020-08-18:
Thanks @jcarpent @jmirabel  It works well now. The rotation order matters for `rpy`.

closed 2020-08-18 by ddliu365