stack-of-tasks/pinocchio #1282: {boost,std}::is_floating_point is only true for primitive floating point types

aelkazdadi, 2020-08-24:
which excludes user defined (e.g., specializations of `boost::multiprecision::number`) and implementation defined (`__float128` on gcc and clang) floating point types.

used in:
```
math/quaternion.hpp
math/matrix.hpp
multibody/joint/joint-common-operations.hpp
```

jcarpent, 2020-08-24:
Thanks for raising this issue @aelkazdadi. Could you elaborate a bit more?

aelkazdadi, 2020-08-24:
code that calls `pinocchio::integrate` on vectors of `T = multiprecision::number<...>` won't compile since it has to instantiate `pinocchio::quaternion::internal::quaternionbase_assign_impl<T, boost::is_floating_point<T>::value>`

`boost::is_floating_point<T>::value` happens to be false for non primitive types, so this doesn't work in this case

mre:
```cpp
#include <pinocchio/algorithm/joint-configuration.hpp>
#include <boost/multiprecision/cpp_bin_float.hpp>

int main() {
  using T = boost::multiprecision::cpp_bin_float_100;
  pinocchio::ModelTpl<T> m;
  Eigen::Matrix<T, Eigen::Dynamic, 1> q;
  pinocchio::integrate(m, q, q, q);
}
```

aelkazdadi, 2020-08-24:
one solution would be to specialize `boost::is_floating_point`, which is a bit intrusive. or we could define and use our own trait that inherits from `boost::is_floating point`, and allow specializing it it for float-like types.

jcarpent, 2020-08-24:
I would suggest adding something like:
```c++
namespace pinocchio
{
template<typename T>
struct is_floating_point : boost::is_floating_point<T>
{}
}
```
and to overload it for specific types.

jcarpent, 2020-08-24:
@aelkazdadi Could you handle it?

jcarpent, 2020-08-24:
Thanks a lot.

jcarpent, 2020-08-29:
Solved by #1283

closed 2020-08-29 by jcarpent