stack-of-tasks/pinocchio #1286: Mass of root link ignored in computeTotalMass?

wxmerkt, 2020-08-31:
Spotted by @oliwiermelon

Is there a reason the mass of the root link is ignored in the `computeTotalMass` and `computeSubtreeMasses` algorithms?

E.g. here the iteration starts with the 1st inertia and not the 0th:

https://github.com/stack-of-tasks/pinocchio/blob/820d0f85fbabddce20924a6e0f781fb2be5029e9/src/algorithm/center-of-mass.hxx#L21

jcarpent, 2020-08-31:
The joint 0 corresponds to the universe. In other words, its mechanical and physical properties are not taken into accounts for all the algorithms such as CBRA, RNEA, etc.
To comply with this setting, we decided to not account for it when computing the total mass of the system, because this mass is not moved by the system when performing motions.

It is then easy to add the universe mass if you need after calling this algorithm. What do you think @wxmerkt?

wxmerkt, 2020-08-31:
I had a suspicion that some algorithm places an assumption on skipping the 0th joint, thanks for the explanation!

I realised that the issue was that we did not specify `JointModelFreeFlyer()` upon creation. Hence, everything works as expected :-).

closed 2020-08-31 by wxmerkt

jcarpent, 2020-08-31:
Sounds perfect. Thanks for the feedback.