stack-of-tasks/pinocchio #1287: Computation of the combined, full-body inertia (SRBD)

oliwiermelon, 2020-08-31:
Hi, 

I am trying to use Pinocchio to calculate the combined, full-body inertia of an articulated body which accounts for the floating base and the end-effectors, in their nominal configuration from a URDF for a Single Rigid Body Dynamics (SRBD) model.

I am aware of the CRBA algorithm which produces the joint-space inertia matrix (_n_  x _n_). I am looking for a function/approach that:
- that transforms it into the reference frame of the base and combines it with its inertia matrix (3 x 3), or
- returns the full-body inertia matrix (3 x 3).

jcarpent, 2020-08-31:
Hi @oliwiermelon,

The best solution would be:

1. call `crba(model,data,q)`
2. retrieve the total spatial inertia with data.Ycrb[1], which is expressed in the base coordinate system (assuming you are working with a free floating system).
3. data.Ycrb[1].inertia() is the quantity you are looking for.

oliwiermelon, 2020-08-31:
Thank you @jcarpent, it is exactly what I needed!
_Note:_ `data.Ycrb[1].inertia`

closed 2020-08-31 by oliwiermelon

jcarpent, 2020-08-31:
> _Note:_ `data.Ycrb[1].inertia`

In Python ;). In C++, `data.Ycrb[1].inertia()`.