stack-of-tasks/pinocchio #1291: Are joint friction parameters from URDF laoded to the robot model?

rshum19, 2020-09-13:
Hi I have a similar question to #1147.  I'm trying to simulate a 7DOF robot arm and want to include joint friction and damping. In the URDF one can define the tag 
```
<joint>
    <dynamics damping="1.0" friction="1.0">
</joint>
```

I'm wondering if these parameters are used at all when loading the model? 

If this is not the case, then would the best way to include joint friction be by adding a joint friction model to the joint torque vector (tau) before computing the forward the dynamics like shown below?
``` 
tau = friction_model(q,v);
ddq = pinocchio::aba(model, data, q, v, tau);
```

thanks,

jcarpent, 2020-09-13:
For the damping term, which affects the torque proportionally to the joint angular velocity, your trick does the job.
For handling the friction, it is a bit harder, as you have to solve a linear complementary problem: 
`tau < friction is orthogonal to ddq > 0`

jcarpent, 2020-09-13:
What we may add, is the automatic loading of these parameters in the Model when parsing the URDF file.

rshum19, 2020-09-13:
thank you for the fast response! 

I see, so adding damping can be easily done but friction requires more effort. I will look more into friction model and adding friction. Your response clarifies my question about what gets parsed from the URDF, so I will close this issue.

closed 2020-09-13 by rshum19

gfadini, 2020-10-01:
Hi, I see that this issue has been closed, nonetheless I think that having those parameters parsed from the URDF automatically may be useful for some applications.
In my specif case I can use the coulomb friction and viscous damping by defining them though crocoddyl, but I guess that since they actually are physical parameters of the system they may be loaded in the model itself. What do you think about this, would this be relevant to other people as well?

jcarpent, 2020-10-01:
Yes, this was my remark.
The new version of Pinocchio 3.x should include them. But, even with these values, the features of Crocoddyl will not allow you to handle correctly the friction, only the damping.

gfadini, 2020-10-01:
Thanks for the answer @jcarpent , that would be a nice addition to the next version of pinocchio. I agree with you that including the static friction properly in crocoddyl won't be straightforward

cmastalli, 2020-10-01:
@jcarpent we could boost the integration is you provide a roadmap with a rough description of the API.

jcarpent, 2020-10-01:
I'm lacking of time for that. Sorry.

cmastalli, 2020-10-01:
> I'm lacking of time for that. Sorry.

No problem at all, thanks :)

Kongx231, 2021-10-06:
I see that damping shows up in the model parameter in Pinocchio3, but doesn't seem to be applying damping forces when applying forward dynamics. That's okay if its still not supported, I can use the solution from above. I just wanted to double check.

jcarpent, 2021-10-06:
Not yet, this will be present in the final version ...
Sorry for the delay.