stack-of-tasks/pinocchio #1292: Obtaining contact forces from torques for a given joint-space configuration

oliwiermelon, 2020-09-15:
Hi,

**Desciption**
I am trying to determine the maximum contact forces at the end-effectors from peak torque values for the nominal joint-space configuration. At least one of the x, y, z components of the forces is an order of magnitude higher than expected. I hope you can identify the flaw in my approach which uses the relationship below (it assumes `J` is a square matrix) to compute the force vector:
```
tau = J^T * F
inv(J^T)*tau = F
```

**Attempt**
Here is what I have attempted:

I load a floating-base quadruped from a URDF:
```
model    = pin.buildModelFromUrdf(urdf_path, pin.JointModelFreeFlyer())
data     = model.createData()
```
I provide `q0`, the nominal state vector (19x1) with the base at `{x, y, z, qx, qy, qz, qw} = {0, 0, 0, 0, 0, 0, 1}` (this puts the `base` frame at the `world` frame). Hence, I obtain the `LF_FOOT` body Jacobian (6x18) w.r.t. `WORLD`:
```
j_full = pin.computeFrameJacobian(model, data, q0, model.getFrameId("LF_FOOT"), pin.ReferenceFrame.WORLD);
```
I extract the square part of the Jacobian corresponding to `LF_FOOT` (3x3):
```
j = j_full[0:3, 6:9]
```
Finally, I compute `f`, the force vector (3x1):
```
jt = j.transpose()
f = pinv(jt).dot(tau)
```

**Question**
Is this the correct use of `computeFrameJacobian()`? Should I be using something else?

jcarpent, 2020-09-15:
You want to extract the contact forces expressed at the LF_FOOT level. 
For this case, the reference frame you provide is not the correct one, meaning that WORLD refers to the origin of the world as if it was attached to the LF_FOOT frame.
In your case, you should rather use LOCAL_WORLD_ALIGNED convention, which corresponds to the Frame cenetred on the LF_FOOT frame but with axes aligned with the WORLD axes.

oliwiermelon, 2020-09-15:
Thank you @jcarpent for a very prompt reply. The number make much more sense after changing the reference frame to `LOCAL_WORLD_ALIGNED`!

closed 2020-09-15 by oliwiermelon