stack-of-tasks/pinocchio #1293: CoM Position and Jacobian Frame

mrsp, 2020-09-15:
Hello 

when you call     
pinocchio::centerOfMass(model, data, q);
pinocchio::jacobianCenterOfMass(model, data, q);

With respect to which frame is the CoM position and the CoM Jacobian computed? Is it i.e. the "base_link"?

Thanks in advance

jcarpent, 2020-09-15:
The center of mass and the related Jacobian are computed w.r.t the WORLD frame. This is written in the C++ doxygen documentation: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#aa68008ef776e6959af9884eff4514633

jcarpent, 2020-09-16:
@mrsp Does it answer your question? If yes, could you close this issue then?

mrsp, 2020-09-16:
yes it does. 

Additionally, if you also have any documentation on all the frames used in Pinocchio  kindly share it with me!

Thanks

jcarpent, 2020-09-16:
The frame are all mentionned in the ReferenceFrame enum. I will try to uptate the documentation to reflect their meaning.

closed 2020-09-16 by jcarpent