stack-of-tasks/pinocchio #1294: compute center of mass acceleration

WangKeAlchemist, 2020-09-17:
Hi,

I want to use Pinocchio to compute center of mass acceleration in world frame, so I use `pin::forwardKinematics(model,data,q); acc_x = data.acom[0][0];` in C++, for each iteration. However later I realize that the acceleration I get is always 0, have I done anything wrong?

Thanks!

jcarpent, 2020-09-17:
No, you should just call the function to compute it, which is `centerOfMass(model,data,q,v,a)`

jcarpent, 2020-09-17:
By the way, `centerOfMass` is already calling `forwardKinematics`

jcarpent, 2020-09-17:
If you want to compute the CoM from the current values of the kinematics stored in Data, you should just call:
`centerOfMass(model,data,2)` where `2`means ACCELERATION level (1 means VELOCITY and 0 means POSITION).

olivier-stasse, 2020-09-17:
> If you want to compute the CoM from the current values of the kinematics stored in Data, you should just call:
> `centerOfMass(model,data,2)` where `2`means ACCELERATION level (1 means VELOCITY and 0 means POSITION).

Maybe an enum would be more informative ?

jcarpent, 2020-09-17:
Yes, I do agree. This is what I realized after writing this answer.

WangKeAlchemist, 2020-09-17:
Hi @jcarpent @olivier-stasse, your answer is very helpful and I just tried that. What I find is: if I use `centerOfMass(model,data,q,v)` only, then the resultant center of mass acceleration is still 0. That is probably because I don't provide joint acceleration (I cannot get joint acceleration directly from my simulation and can only get it by numerical differentiation). But when I use `centerOfMass(model,data,2)` and it does work! 
Just out of curiosity,  does `centerOfMass(model,data,2)`  use numerical differentiation to compute CoM acceleration given joint acceleration is not provided?

jcarpent, 2020-09-17:
You have forgotten an argument: centerOfMass(model,data,q,v,**a**)

WangKeAlchemist, 2020-09-17:
> You have forgotten an argument: centerOfMass(model,data,q,v,**a**)

Right, after adding **a** it works well, thanks!

closed 2020-09-17 by jcarpent