stack-of-tasks/pinocchio #1295: Using pinocchio with catkin

jakewelde, 2020-09-17:
Hi, I'm trying to build against pinocchio, which I installed by including it as a <depend> in the package.xml of a package in the catkin workspace of my ROS codebase. I then installed pinocchio using  

```rosdep install --from-paths src --ignore-src -r -y```

When I try to run catkin_make, I get the error

```
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Found package configuration file:

    /opt/ros/noetic/share/pinocchio/cmake/pinocchioConfig.cmake

  but it set pinocchio_FOUND to FALSE so package "pinocchio" is considered to
  be NOT FOUND.

```

Could anyone possibly help me understand what I'm doing wrong? My understanding of the build system is limited so I really appreciate any help. Thanks!

jcarpent, 2020-09-17:
@wxmerkt Could you please handle this issue?

wxmerkt, 2020-09-17:
Hi @jakewelde,
Happy to take a look at this. Can you please post the excerpt of the CMakeLists.txt where you find Pinocchio and how you link against it?

Please note you need to find the package in CMake as `find_package(pinocchio REQUIRED)` and not via catkin. You can then link against it using `target_link_libraries(${PROJECT_NAME} pinocchio::pinocchio)`.

jakewelde, 2020-09-17:
Thanks so much for your quick reply! I believe I've updated my CMakeLists.txt according to your instructions. It's included below. 

> cmake_minimum_required(VERSION 2.8.3)
> project(aerial_manipulator_visualization)
> 
> if(NOT CMAKE_BUILD_TYPE)
>   set(CMAKE_BUILD_TYPE RelWithDebInfo)
> endif()
> 
> set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -Wall")
> 
> find_package(catkin REQUIRED COMPONENTS geometry_msgs sensor_msgs nav_msgs roscpp aerial_manipulator_platforms
>   topic_tools)
> 
> 
> **find_package(pinocchio REQUIRED)**
> find_package(Eigen3 REQUIRED)
> 
> 
> include_directories(include ${catkin_INCLUDE_DIRS} ${EIGEN3_INCLUDE_DIR})
> 
> 
> catkin_package(
>   INCLUDE_DIRS
>   LIBRARIES
>   CATKIN_DEPENDS geometry_msgs sensor_msgs nav_msgs roscpp topic_tools aerial_manipulator_platforms
>   DEPENDS Eigen3)
> 
> include_directories(${catkin_INCLUDE_DIRS})
> 
> add_executable(aerial_manipulator_visualization src/aerial_manipulator_visualization.cpp)
> add_dependencies(aerial_manipulator_visualization ${catkin_EXPORTED_TARGETS})
> target_link_libraries(aerial_manipulator_visualization ${catkin_LIBRARIES})
> 
> **target_link_libraries(aerial_manipulator_visualization pinocchio::pinocchio)**
> 
> 
> install(TARGETS ${PROJECT_NAME}
>   ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
>   LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
>   RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
> )
> 
> install(DIRECTORY launch/ DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/launch)
> install(DIRECTORY rviz/ DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/rviz)
>

jakewelde, 2020-09-17:
Not sure if this is helpful, but based on this issue #1261 I ran the command

```pkg-config --libs pinocchio```

which gave me the result

```Empty package name in Requires or Conflicts in file '/opt/ros/noetic/lib/pkgconfig/pinocchio.pc```

jakewelde, 2020-09-17:
I've also tried to install pinocchio according to the standard instructions [here](https://stack-of-tasks.github.io/pinocchio/download.html), but when I run

```sudo apt install -qqy robotpkg-py27-pinocchio  # Adapt your desired python version here```

I get the error

```E: Unable to locate package robotpkg-py27-pinocchio```

(I am running ROS Noetic on Ubuntu 20.04)

wxmerkt, 2020-09-17:
Hi @jakewelde,
I put together a sample project showing how to include and use Pinocchio in a catkin project: https://github.com/wxmerkt/pinocchio_ros_example

Note, this buildtests against the ROS-distributed version in Noetic as well - and this worked: https://travis-ci.com/github/wxmerkt/pinocchio_ros_example/jobs/386377260

Please note that once you start mixing robotpkg and ROS distributed Pinocchio things become quite complex (you need to be careful about the paths etc.). Can you catkin clean and make? (Note: both locally and on CI we only use `catkin build` from catkin tools rather than `catkin_make` in case that makes a difference).

jakewelde, 2020-09-17:
Thanks so much for your detailed help!

After doing a `make clean` I was able to build the example package and then also add the code to my other package and build it. I think you're spot on that my several attempts to install pinocchio through different means confused things. 

However, when I try to include both <ros/ros.h> with the example code you included for Pinocchio, I get template errors like
` error: wrong number of template arguments (21, should be at least 0)` related to the Pinocchio code.
I can post more detailed errors but don't want to spam you. Do you think this could be related to using catkin_make instead of catkin build? I can try switching over but will have to try and understand the details of the differences.

jakewelde, 2020-09-17:
I just saw your comment on #1056 regarding include order and was able to build after including <ros/ros.h> after the pinocchio includes!

jakewelde, 2020-09-17:
Thanks again for your prompt assistance on this, really appreciate it!

closed 2020-09-17 by jakewelde

jcarpent, 2020-09-17:
Thanks @wxmerkt for a precise and concise answer.