stack-of-tasks/pinocchio #1297: Derivative of center of mass Jacobian

jakewelde, 2020-09-18:
Hi, I am wondering if you could help direct me to the best way to use Pinocchio to compute the time derivative of the center of mass jacobian, i.e. for center of mass x, since x' = J v, then x'' = J v' + J' v and I want to compute J' (I am already computing the rest using jacobianCenterOfMass). Is there an analytical implementation I haven't been able to find, or could you offer a tip for where to look in the documentation to compute this via autodiff? Many thanks for your excellent library and helpful support.

jcarpent, 2020-09-18:
`computeCentroidalMapTimeVariation` will compute `dAg/dt`, the time variation of the centroidal momemtum matrix. The 3 first rows do match the center of mass Jacobian time variation.
If you need a dedicated algorithm for computing this quantity, I may implement it.

jakewelde, 2020-09-18:
I think the existing functionality fully satisfies my needs! thank you again.

closed 2020-09-18 by jakewelde

junhyeokahn, 2021-01-21:
@jakewelde @jcarpent Don't you need to divide ```dAg/dt``` by the ```total_mass``` of the robot to achieve a com jacobian dot?

jakewelde, 2021-01-21:
Agreed, this is also a required detail

jcarpent, 2021-01-21:
> @jakewelde @jcarpent Don't you need to divide `dAg/dt` by the `total_mass` of the robot to achieve a com jacobian dot?

Yes, you are correct.

junhyeokahn, 2021-01-21:
@jcarpent @jakewelde  Thanks for verification!