stack-of-tasks/pinocchio #1306: kinematic energy

ddliu365, 2020-10-02:
I have a question about kinematic energy, did I call it correctly?
In the simulation, I did an experiment where the robot contact the ground by knee and it fall freely to the ground. The calculated increased kinetic energy and reduced potential energy have a big difference, which is weird. Since there is no other force to do work and we almost have nothing energy input from motors. 
Can you help check my energy coding?
```
peArr[0,i] = se3.computePotentialEnergy(self.env.pinocchio_robot.model, self.env.pinocchio_robot.data, q)
keArr[0,i] = se3.computeKineticEnergy(self.env.pinocchio_robot.model, self.env.pinocchio_robot.data, q, dq)
```

[image]

jcarpent, 2020-10-03:
First of all, we would really appreciate to close issues that you have opened so far (e.g. #1302, with at least a confirmation message or even better, with some thanks). We are trying to provide feedback to our users as best as we can, we may expect a similar feedback from them.

Concerning your issue, did you check on a minimal but relative simple example as an inverted pendulum to check the conservation of energy of your system?
In the present context, it seems you are handling collisions and so on, so there are no reasons to observe the conversation of energy in this context. In addition to explain your context, it would be nice to also provide your snippet of your code.
Last but not least, your title is incorrect: you are talking about **kinetic** energy.

ddliu365, 2020-10-03:
The collision actually occurred after a falling period. But before the contact during the falling period, the kinetic energy is not equal to potential energy and they have a big difference. 

Right now I don't have a inverted pendulum model constructed for `pinocchio`.

The simulation environment is bullet physics. For kinetic energy calculation, did I make any mistake? I just found the function in the document. 

Thanks.

jcarpent, 2020-10-03:
> The collision actually occurred after a falling period. But before the contact during the falling period, the kinetic energy is not equal to potential energy and they have a big difference.

Why the kinetic energy and potential energy should be equal? The physics just tells us that the total mechanical energy is preserved when an object is falling. This is exactly what your curves show.

ddliu365, 2020-10-03:
Sorry, I mean the reduced part of potential energy is converted to increased part of kinetic energy. From the curve, the these two parts have a big difference. 
The potential energy reduced about 2.8 J but only kinetic energy only increased about 0.45 J. 
This is the reason why I doubt I make a mistake while writing the code.

jcarpent, 2020-10-03:
It would better to ask people that are making Bullet. I would check if the models are correct and also if the gravity fields match.

ddliu365, 2020-10-03:
Okay. If the `pinocchio` API I am writing is fine I will check it with `bullet`. Thank you very much.

closed 2020-10-03 by ddliu365