stack-of-tasks/pinocchio #1307: Fail to compile with scalar type = casadi::SX

matteoparigi, 2020-10-05:
Hi all, 
pinocchio (latest devel -> aa39638ffb079078c805b562f55f71cdb5fedb13) fails to compile when trying to cast an existing model to the scalar type `casadi::SX`. Attached the error by GCC 7.4

```
from myws/external/casadi_kin_dyn/src/casadi_kin_dyn.cpp:6:
myws/build/install/include/pinocchio/spatial/se3-tpl.hpp: In instantiation of ‘static void pinocchio::internal::cast_call_normalize_method<pinocchio::SE3Tpl<_Scalar, _Options>, NewScalar, Scalar>::run(T&) [with T = pinocchio::SE3Tpl<casadi::Matrix<casadi::SXElem>, 0>; Scalar = double; int Options = 0; NewScalar = casadi::Matrix<casadi::SXElem>]’:
myws/build/install/include/pinocchio/spatial/se3-tpl.hpp:296:73:   required from ‘pinocchio::SE3Tpl<NewScalar, Options> pinocchio::SE3Tpl<Scalar, Options>::cast() const [with NewScalar = casadi::Matrix<casadi::SXElem>; _Scalar = double; int _Options = 0]’
myws/build/install/include/pinocchio/multibody/model.hpp:242:32:   required from ‘pinocchio::ModelTpl<NewScalar, Options, JointCollectionTpl> pinocchio::ModelTpl<Scalar, Options, JointCollectionTpl>::cast() const [with NewScalar = casadi::Matrix<casadi::SXElem>; _Scalar = double; int _Options = 0; JointCollectionTpl = pinocchio::JointCollectionDefaultTpl]’
myws/external/casadi_kin_dyn/src/casadi_kin_dyn.cpp:111:42:   required from here
myws/build/install/include/pinocchio/spatial/se3-tpl.hpp:351:9: error: could not convert ‘casadi::operator>(const casadi::Matrix<casadi::SXElem>&, const casadi::Matrix<casadi::SXElem>&)(Eigen::GenericNumTraits<T>::epsilon() [with T = casadi::Matrix<casadi::SXElem>; Eigen::GenericNumTraits<T>::Real = casadi::Matrix<casadi::SXElem>]())’ from ‘casadi::Matrix<casadi::SXElem>’ to ‘bool’
         if(pinocchio::cast<NewScalar>(Eigen::NumTraits<Scalar>::epsilon()) > Eigen::NumTraits<NewScalar>::epsilon())
         ^~
```

Thank you for you time,
Matteo

jcarpent, 2020-10-05:
This is a bit weird, because casadi is compiling well on our CI: https://github.com/stack-of-tasks/pinocchio/runs/1202099992
Could you run the Pinocchio tests that involve Casadi (with ctest -R test-cpp-casadi --output-on-failure)?

matteoparigi, 2020-10-11:
Thanks for your reply. I did and all the tests passed.  Yet I still get the error when compiling, unless I comment line 351 in se3-tpl.hpp

jcarpent, 2020-10-11:
I think you just forgot to include the file: `pinocchio/autodiff/casadi.hpp` as done here for instance: https://github.com/stack-of-tasks/pinocchio/blob/820d0f85fbabddce20924a6e0f781fb2be5029e9/unittest/casadi-basic.cpp#L5

matteoparigi, 2020-10-11:
Yes, you are right. Thanks a lot!

jcarpent, 2020-10-11:
Perfect. Good news!

closed 2020-10-11 by jcarpent