stack-of-tasks/pinocchio #1316: Normalization of SpecialEuclidean and SpecialOrthogonal joint configuration where input has zero norm.

gabrielebndn, 2020-10-13:
Normalization methods SpecialEuclidean and SpecialOrthogonal rely on Eigen's `normalize()` method. According to [the method's doc](https://eigen.tuxfamily.org/dox/classEigen_1_1MatrixBase.html#ad16303c47ba36f7a41ea264cb26bceb6), if the vector's norm is too small, then it is left unchanged. Because of this, if the input vector contains all zeroes at the place of quaternions/continuous joints, the output of Pinocchio's `normalize` is _not_ normalized.

This is unexpected behaviour. A possible solution would be normalizing to the neutral configuration for the specific joint where Eigen's normalization fails. Would you agree on this? I can handle it as soon as I have time.

Additionally, it would be helpful to add a method called `isNormalized` for joint configurations (and, more in general, `isValid`, checking normalization _and_ bounds) (for things like checking user's input and such).

jcarpent, 2020-10-13:
> This is unexpected behaviour. A possible solution would be normalizing to the neutral configuration for the specific joint where Eigen's normalization fails. Would you agree on this? I can handle it as soon as I have time.

It might be dangerous I would say, leading to some discontinuities, everything leading to the neutral element. I would prefer to avoid that, and I do think @jmirabel may provide his feedback on it too.

> Additionally, it would be helpful to add a method called `isNormalized` for joint configurations (and, more in general, `isValid`, checking normalization _and_ bounds) (for things like checking user's input and such).

These new helper functions are more than welcome. Please go ahead.

jmirabel, 2020-10-13:
FYI,

- the `IsNormalized` exists in *hpp-pinocchio*. Copy-paste may (or may not) be simpler than starting from scratch.
- I think the `normalize` was introduced to overcome rounding errors and the zero quaternion example is more likely to be a programming mistake than a rounding error. I think it is better to have one tool for each goal. That being said, I agree the method name may be misleading (as it is in Eigen).

jmirabel, 2020-10-13:
We may decide to do as it is done in Eigen: state the small norm issue in the doc.

jcarpent, 2020-10-13:
Fine with me. Thanks @gabrielebndn for handling this.

gabrielebndn, 2020-10-13:
> > * Add `isValidConfiguration` (with precision for normalized, but hard check for the limits)
> 
> You'll have to pass the limits as arguments.

The idea is that there is no need to implement `isValidConfiguration` joint per joint. It can be done at the model level, and checked element-by-element. So I was thinking of using the internal `lowerPositionLimit` and `upperPositionLimit`. I can do it with custom limits too, of course.

jcarpent, 2020-10-13:
Providing the two signatures would be nice indeed.

jmirabel, 2020-10-13:
In which class do you want to implement the `isValidConfiguration` function ? Lie group classes do not have limits as member attributes.

jcarpent, 2020-10-13:
The guess is to perform that at the model configuration level for the whole using the difference function.

gabrielebndn, 2020-10-13:
> In which class do you want to implement the `isValidConfiguration` function ? Lie group classes do not have limits as member attributes.

No class. I am not planning to implement it for lie groups. Just a free function for the model. Basically (just a sketch):
```c++
bool isValidConfiguration(Model model, VectorXd q, prec)
{
  if(!isNormalized(model, q, prec))
    return false;

  for(int i=0; i<=model.nq; i++)
  {
    if(q[i] <= model.lowerPositionLimit[i] || q[i] >= model.upperPositionLimit[i])
      return false;
  }

  return true;
}
```

jcarpent, 2020-10-13:
Please do it for Lie groups. It is just a matter of using Pinocchio::difference.

florent-lamiraux, 2020-10-14:
> We may decide to do as it is done in Eigen: state the small norm issue in the doc.

What about an assert, or an exception for those who never compile in debug mode ?

gabrielebndn, 2020-10-14:
> Please do it for Lie groups. It is just a matter of using Pinocchio::difference.

Not sure about it. First of all, the concept of bound for Lie Groups is dubious at best as far as I understand, second of all I am really not sure on how to implement it with quaternions (and it can be confusing for `SO(2)` too, what about `\pi - \delta <= \theta <= \pi + \delta`)? The truth is the bounds on the quaternion part of the configuration have never had a lot of sense and we only have them for ease of implementation. @jmirabel what do you think?

I think that for the bounds I'll stick to the element-by-element comparison performed at the model level. If you think it is an inadequate implementation, then I'll leave it to you, it was just a suggestion :slightly_smiling_face:

jcarpent, 2020-10-14:
@gabrielebndn, you just need to use the correct operations, and pinocchio::difference will be transparent in your implementation. This is the only correct operator. In the case of quaternion parts, the limit is a bit above the one value for each component, and it won't affect the check. Test it. This is my only but main concern.

jmirabel, 2020-10-14:
> Not sure about it. First of all, the concept of bound for Lie Groups is dubious at best as far as I understand, second of all I am really not sure on how to implement it with quaternions (and it can [...]

As you said, checking the bound of a unit quaternion or a unit complex is meaningless. Two useful things:
- a light check that ensure the coefficient are within [-1, 1], with a tiny epsilon.
- check the normalization.
For both, the target is to ease debugging.

> Please do it for Lie groups. It is just a matter of using Pinocchio::difference.

@jcarpent I think I don't understand what you mean. What do you want to do with `difference` ?

jcarpent, 2020-10-14:
> @jcarpent I think I don't understand what you mean. What do you want to do with `difference` ?

As simple as: `pinocchio::difference(model,q,lb).array() >= 0.`

jcarpent, 2020-10-14:
This is exactly how it works in TSID and other related frameworks.

jmirabel, 2020-10-14:
This won't work for quaternions.

gabrielebndn, 2020-10-14:
> This won't work for quaternions.

That's exactly my point

gabrielebndn, 2020-10-14:
> Two useful things:
> 
>     * a light check that ensure the coefficient are within [-1, 1], with a tiny epsilon.
> 
>     * check the normalization.

@jmirabel, I'll just note that the second point implies the first one. If the quaternion is normalized, then its coefficients are certainly  within [-1, 1], so no need to check it separately.

jmirabel, 2020-10-14:
Yes, but this means you can safely do lb <= q <= ub, assuming lb and ub are properly set.

gabrielebndn, 2020-10-14:
> Yes, but this means you can safely do lb <= q <= ub, assuming lb and ub are properly set.

Of course, that was the idea

jmirabel, 2020-10-14:
About the function name, I think `isValidConfiguration` is not precise enough. One may think you do collision checking too.
Not sure exactly of what function would suit it more:
- `isConfigurationWithinBounds`,
- `isWithinBounds`
- `isConfigInConfigSpace`

The unit-norm check being somehow equivalent to the standard joint bounds, I like the second one.

jmirabel, 2020-10-14:
`isValidConfiguration` isn't very short (and makes the code stutter).

From the developer perspective, I judge
```cpp
pinocchio::isWithinBounds(model, config);
```
more readable than
```cpp
pinocchio::isValidConfiguration(model, config);
```

jcarpent, 2020-10-14:
There a two thing:
`isValidConfiguration` is calling `isWithinBounds` + `isNormalized`.

gabrielebndn, 2020-10-14:
> `isValidConfiguration` isn't very short (and makes the code stutter).
> 
> From the developer perspective, I judge
> 
> ```c++
> pinocchio::isWithinBounds(model, config);
> ```
> 
> more readable than
> 
> ```c++
> pinocchio::isValidConfiguration(model, config);
> ```

The idea was that `isValidConfiguration` would check that it is normalized _and_ within bounds. BTW, I am starting to reconsider the need of checking the bounds, after all it is a one-liner in C++ :
```
((q-lowerBound).array() >= 0).all() && ((upperBound-q).array() >= 0).all()
```

jcarpent, 2020-10-14:
Yes, but you need to know Eigen.

gabrielebndn, 2020-10-14:
Otherwise, I can just implement `isWithinBounds` and let users do
```
isNormalized(q) && isWithinBounds(model, q);
```
Doesn't need any advanced Eigen knowledge ;)

jcarpent, 2020-10-14:
I'm not against it, and should be easy.

gabrielebndn, 2020-10-14:
Perfect

gabrielebndn, 2020-10-22:
Closed by #1324

PR #1324 (2020-10-22): Add isNormalized
Closes #1316.

I have implemented `isNormalized` for lie groups and for the joint configuration, with bindings.

I have very limited time, so I think I will let the bounds checks go for the time being. The normalization check has much more value and it is much more difficult to implement for an inexpert user, so I think this contribution is already important per se.
closed 2020-10-22 by gabrielebndn