stack-of-tasks/pinocchio #1318: Display issues in wheels

nim65s, 2020-10-13:
Hi again,

To check #1317, I created a docker that install everything from pip, and then try to run [a jupyter notebook with talos saying hi](https://github.com/MeMory-of-MOtion/summer-school/blob/master/setup/load_talos.ipynb).

This is failing both with GepettoViewer and MeshcatViewer. If I have a `pinocchio.pinocchio_pywrap.GeometryObject` and try to access its `geometry` attribute, as it is done [here](https://github.com/stack-of-tasks/pinocchio/blob/17c479e2fe91905122713643f489feb8ddf135a3/bindings/python/pinocchio/visualize/meshcat_visualizer.py#L103), I get:
```
No to_python (by-value) converter found for C++ type: boost::shared_ptr<pinocchio::fcl::FakeCollisionGeometry>
```

To reproduce this (with meshcat):
```
docker run --rm -p 8888:8888 -p 7000:7000 -it gepetto/utils:wheel-3.9
```

jcarpent, 2020-10-13:
This is not loading hppfcl Python bindings in fact.

jcarpent, 2020-10-26:
Is the problem fixed?

nim65s, 2020-10-26:
Not yet, I'm spending too much time on the robot, sorry

jcarpent, 2021-01-04:
The problem here is that the current project is not compiled with HPP-FCL dependency.

closed 2021-01-04 by jcarpent