stack-of-tasks/pinocchio #1320: Pinnochio as subdirectory

lpdon, 2020-10-14:
Hello. I would like to add pinocchio as subdirectory in a project, without using catkin or robotpkg. I can build it from source independently without problems i.e as a single CMake project but some CMake variables just seem broken when pinocchio is added as a subdirectory (e.g the .cmake includes in the base CMakeLists.txt assume the project is the root). I tried to fix some of the CMake files, but so many problems arise after each fix (first with the PROJECT_JRL_ scope, then mathjax, then LOGGING_WATCHED_VARIABLES etc) that I just gave up. What would one actually need to setup to use it properly as subdirectory?

jcarpent, 2020-10-14:
Hi and thanks for using Pinocchio.

We are changing the way the CMake submodule is working.
I think @nim65s is the right person to discuss this point.

@lpdon Could you provide an example in such a way we can itterate on it?

lpdon, 2020-10-14:
Hi, thanks for the feedback. I did a really quick&dirty just to get it to build and move forward. What I changed in the CMake files can be seen in https://github.com/lpdon/pinocchio/ and https://github.com/lpdon/jrl-cmakemodules i.e basically using CMAKE_CURRENT_LIST_DIR to find the .cmake files and CACHE INTERNAL for some variables that were used outside of the the file which defined them in the first place.

jcarpent, 2021-01-06:
@lpdon I've just merged a new PR #1365 into the devel branch where I've included some of your suggestions. It works well and I also add a test over it inside the CI.

Thanks for this very useful feature.

closed 2021-01-06 by jcarpent