stack-of-tasks/pinocchio #1322: Compiling pinocchio related libraries is too time consuming

Wenchun6, 2020-10-15:
It seems usually get stuck when compiles a library need to link to Pinocchio
.so on a poor performance cpu

jcarpent, 2020-10-15:
We have started to work on this topic with https://github.com/stack-of-tasks/pinocchio/pull/1246, but it requires some additional work to be merged. Sorry for the current limitations. 
On which algorithms do you have troubles in particular?

Wenchun6, 2020-10-15:
> We have started to work on this topic with https://github.com/stack-of-tasks/pinocchio/pull/1246, but it requires some additional work to be merged. Sorry for the current limitations. 
> 
> On which algorithms do you have troubles in particular?
Well, everything is ok actually, I just found it too time and memory consuming when compiled on my embedded computer

jcarpent, 2020-10-15:
OK. If it is a strict limitation for you, you are invited to help us with this task. Otherwise, I close this issue and relates it to the ongoing PR.

closed 2020-10-15 by jcarpent

cmastalli, 2021-02-02:
I think an issue should not be closed before the code is merged. In particular it is important in this case as the PR is still pending from many months