stack-of-tasks/pinocchio #1326: getKKTContactDynamicMatrixInverse without calling forwardDynamics/impulseDynamics

mayataka, 2020-10-21:
Dear all,

First of all, I would like to thank you very much for your wonderful repository. I heavily use pinocchio for optimal control of robots.

I have a question and a wish on the getKKTContactDynamicMatrixInverse: can we call getKKTContactDynamicMatrixInverse without calling forwardDynamics nor impulseDynamics beforehand?
This seems possible since we can compute the inverse of the KKT matrix of contact dynamics as long as we have the joint inertia matrix and the contact Jacobian, e.g, by computed by calling crba and getFrameJacobian.
If we cannot, it is very helpful to add such function to get the inverse of KKT matrix of contact dynamics.

Thank you.

jcarpent, 2020-10-21:
Hi @mayataka,

Thanks for your thanks.
Your request is indeed very easy to handle. I think you should have the right skills to handle it.
On my side, I may give a try this morning.

Best,
Justin

mayataka, 2020-10-21:
Dear @jcarpent 

Thanks for replying.
OK, I'll try on my own. I guess this may be possible just by modifying forwardDynamics.
Thanks a lot.

Regards,
Sotaro Katayama

closed 2020-10-21 by mayataka

jcarpent, 2020-10-21:
@mayataka I'm on it with some side mofiications (with fixes). Just for next time, I think it would be easy for you to handle such kind of request, where I can provide feedback.

Thanks a lot for your involvement.