stack-of-tasks/pinocchio #1330: Unexpected behavior of log6 with infinity values

cmastalli, 2020-10-27:
Dear developers,

Using the following code:
```python
import pinocchio
import numpy as np

M = pinocchio.SE3(np.eye(3), np.array([0.1, 0.1, 0.1]))
Mo = pinocchio.SE3(np.eye(3), np.array([np.infty, 0.1, 0.1]))

print pinocchio.log6(Mo.inverse() * M)
```
I obtained these values:
```bash
  v = -nan -nan -nan
  w = 0 0 0
```

But if my understanding is correct, we should get
```bash
  v = -nan 0 0
  w = 0 0 0
```

Additionally, if I change `Mo` for a high value, i.e.
```python
Mo = pinocchio.SE3(np.eye(3), np.array([1e9, 0.1, 0.1]))
```
I obtained what I expect, i.e.
```bash
  v = -1e+09      0      0
  w = 0 0 0
```

As a closing comment, I did not try other operators, but they need to be revised as well.

Carlos

jcarpent, 2020-10-27:
Hi Carlos,

I think you should read first the related code to understand the way the values are propagated through the computational graph, as stated here:
https://github.com/stack-of-tasks/pinocchio/blob/17c479e2fe91905122713643f489feb8ddf135a3/src/spatial/log.hxx#L121-L142

As you might see, there is no way to avoid the propagation of NaN numbers within the results.

cmastalli, 2020-10-27:
I see, thanks for the inside.

closed 2020-10-27 by cmastalli