stack-of-tasks/pinocchio #1331: Is there a complete implmentation of tuturial 6 "take a walk"?

tigerpat, 2020-10-31:
I am interested in trying out the "take a walk" tutorial described here https://github.com/stack-of-tasks/pinocchio/blob/master/doc/d-practical-exercises/6-wpg.md
Can anyone point out if there is a complete implementation of it?
Thanks.

jcarpent, 2020-10-31:
I think @nmansard is the right person to answer your question.

nmansard, 2020-11-05:
that is quite an old piece of code that you diged up! The tutorials that are pushed in Pinocchio are quite old, and there is an external project with a renew of them (with a pending task to integrate them inside the main repo of Pinocchio).
@nim65s: you are the one knowing the proper reference of the latest version of the new tutorial. Could you add them to this issue?

Now, I am afraid that we removed the walk from the new version of the tutorial. Although, I think we still have a master class that is using it, so I am going to dig up myself and see if I can find an update of them, with a complete implementation. Stay tuned!

nmansard, 2020-11-05:
Sorry, I checked and did not find any complete implementation of the walk.

jcarpent, 2020-11-05:
@tigerpat I think your contribution is very well come if you think you really need this tutorial to be complete.
Have you tried so far?

jcarpent, 2020-11-10:
I will close this issue as there is no activity. @tigerpat Feel free to reopen it if needed.

closed 2020-11-10 by jcarpent