stack-of-tasks/pinocchio #1333: calculate Jacobian matrix between robot configuration and 2d-reprojected keypoints

khanhha, 2020-11-01:
Hi all, 

Thanks for creating a wonderful dynamics framework.

I would like to ask two questions regarding the usage of Pinocchio for a computer vision project that involves projecting 3d human joints to 2d points given a camera projection matrix.

- My first question is about how to calculate the Jacobian matrix that relates the changes of NV [model.nv] to the reprojected 2d key-points on the image space. 

So far I have used the function __getJointJacobian__ to get a Jacobian matrix of 6x34, where 6, as far as I understood, represents the linear and angular changes of each joint. 34 is NV of the human model. Now assume that I have a jacobian matrix [whose shape is 2x3] for the camera projection equation that relates 3d points to 2d points.  If my understanding is correct, the final Jacobian matrix that I am trying to find should have a shape of 2x34. However, it seems impossible to me to combine the projection Jacobian of shape 2x3 with the Pinocchio Jacobian matrix of shape 6x34.  

`               jac = pin.getJointJacobian(model, data, j_idx, pin.WORLD)`

- My second question is how to how to calculate the error vector on the image space. In 3d, we have both rotation and position for calculating the difference, as shown in the below code, but in image space, I just have 2d keypoints. How can I calculate the error vector in this case?  

`                goal = SE3.Identity()
                goal.translation = tar_loc
                # dummy current SE3 with identity rotation
                cur = SE3.Identity()
                cur.translation = data.oMi[j_idx].translation
                dMi = goal.actInv(cur)
                err = pin.log(dMi).vector
`
I look forward to hearing from you soon.

jcarpent, 2020-11-01:
Hi @khanhha,

Thanks for using Pinocchio.
Your questions are interesting and relate to the field of Visual Servoing, etc.

>     * My first question is about how to calculate the Jacobian matrix that relates the changes of NV [model.nv] to the reprojected 2d key-points on the image space.

For this point, you need to compute an intermediate quantity which is corresponds to the Jacobian that links the motion of the camera, to the motion of a point in the image space. Follows this link https://robotacademy.net.au/lesson/the-image-jacobian to understand it more deeply. And more extensively, there is the Handbook of Robotics: https://hal.inria.fr/hal-01355384/document

>     * My second question is how to how to calculate the error vector on the image space. In 3d, we have both rotation and position for calculating the difference, as shown in the below code, but in image space, I just have 2d keypoints. How can I calculate the error vector in this case?

Via the previous relation that links the camera velocity (or the joint velocity) to the point velocity, you can easily define a relation between you error in the image space and a joint displacement. For computing the error in the image, they should exist various methods coming from Computer Vision that do that (I'm not expert so far, so this is the simplest element that I can provide at this stage).

khanhha, 2020-11-01:
@jcarpent 

Thank you very much for your quick response on a Saturday night. 

The document is very helpful for me.

Just another quick question, can I consider the 6D camera velocity in this case as the 6D joint velocity, as the first dimension of the Pinocchio Jacobian matrix [6x34]?

Does this mean that I can combine the 2x6 Jacobian camera with the 6x34 Pinocchio Jacobian matrix to get the final Jacobian matrix?

Thank you very much.

jcarpent, 2020-11-01:
We are Sunday here today ;)

> Just another quick confirmation because in my case, the camera is static. Can I consider the 6D camera velocity in this case as the 6D joint velocity, as the first dimension of the Pinocchio Jacobian matrix [6x34]?

Yes, that's correct.

> Does this mean that I can combine the 2x6 Jacobian camera with the 6x34 Pinocchio Jacobian matrix to get the final Jacobian matrix?

Yes, this is exactly what you need to do.

khanhha, 2020-11-01:
@jcarpent 
That's awesome. 
Thank you very much for your help. I really appreciate it. 
Will absolutely cite Pinocchio in our work!

jcarpent, 2020-11-01:
Again, thanks for using Pinocchio and thanks in advance to acknowledge our work in your scientific publications. We really appreciate.

I will close this issue as it seems to be solved. Feel free to reopen it if needed.

closed 2020-11-01 by jcarpent