stack-of-tasks/pinocchio #1337: undefined id, idx_q and idx_v of the root joint

khanhha, 2020-11-07:
hi, 
I would like to ask why the fields id, idx_q and idx_v of the root joint of the model, loaded in python are undefined? I would like to use these index fields, but their values are negative.
[image]
Thanks

jcarpent, 2020-11-07:
The root joint as first joint of the kinematic tree starts at index 0.

khanhha, 2020-11-07:
Thanks. I suppose that this is a bug right?

jcarpent, 2020-11-07:
No. This is just non initialized values.

jcarpent, 2020-11-07:
I made a mistake in my first answer. I’ve corrected it.

closed 2020-11-07 by jcarpent

khanhha, 2020-11-07:
but what does this joint mean/store? The joint after it in the list model.joints also have id==1

jcarpent, 2020-11-07:
Index 0 refers to the world. So for instance, data.oMi[0] refers to placement of the world with respect to itself, so should be equal to Identity.
In the same vein, model.joints[0] refers to the joint that links the world to itself. In another word, this is a fixed joint which is not considered in Pinocchio.

khanhha, 2020-11-07:
@jcarpent: I got it. Thank you for your clear explanation. 
Best