stack-of-tasks/pinocchio #1338: getFrameAcceleration inconsistency with 2.5 version

mfocchi, 2020-11-10:
Hi all 

since when I updated from Pinocchio 2.4.7 to 2.5 I start getting this error when I access the linear / angular fields

TypeError: No to_python (by-value) converter found for C++ type: Eigen::Ref<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 0, Eigen::InnerStride<1> >

e.g. 
 `a = pin.getFrameAcceleration(model, data, 1) `
`  print(a.linear)`

jcarpent, 2020-11-10:
You should update eigenpy in fact. Sorry for this fact, it seems that the minimal version is not up to date.

jcarpent, 2020-11-10:
EigenPy >= 2.5.0. How do you install EigenPy/Pinocchio?

jcarpent, 2020-11-10:
This also means that from now on, we are sharing the memory between Numpy and Pinocchio, in other words
```
M = pin.SE3.Random()
M.translation[:] = 1.
```
will set to Ones the translation part of `M`

jcarpent, 2020-11-10:
> Sorry for this fact, it seems that the minimal version is not up to date.

I've checked, and the requested version is up to date.

mfocchi, 2020-11-10:
@jcarpent 

I have version 2.4.0 of EigenPy installed.

I had installed Pinocchio with
` sudo apt-get install robotpkg-pinocchio`
` sudo apt-get install robotpkg-py27-pinocchio (just to be sure)`

when I try to ugrade eigenpy  with 
`sudo apt-get install robotpkg-eigenpy`

I got this strange error:

Get:1 http://robotpkg.openrobots.org/packages/debian/pub xenial/robotpkg amd64 robotpkg-eigenpy amd64 1.4.0r2 [167 kB]
Fetched 167 kB in 0s (606 kB/s)          
(Reading database ... 677691 files and directories currently installed.)
Preparing to unpack .../robotpkg-eigenpy_1.4.0r2_amd64.deb ...
Unpacking robotpkg-eigenpy (1.4.0r2) ...
dpkg: error processing archive /var/cache/apt/archives/robotpkg-eigenpy_1.4.0r2_amd64.deb (--unpack):
 trying to overwrite '/opt/openrobots/lib/pkgconfig/eigenpy.pc', which is also in package robotpkg-py27-eigenpy 2.4.0
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/robotpkg-eigenpy_1.4.0r2_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

jcarpent, 2020-11-10:
You should install using:
`robotpkg-py27-pinocchio` and be sure to uninstall `robotpkg-eigenpy` which is now outdated because of Python 2.7 and 3.x that we are now supporting.

jcarpent, 2020-11-10:
Normally, `robotpkg-py27-eigenpy-2.5.0` should be installed at the same time.

nim65s, 2020-11-10:
@mfocchi : Hi, your `robotpkg-eigenpy` is really old and should be removed. As @jcarpent pointed, the correct one is `robotpkg-py27-eigenpy`, as you can now choose the python version you want to use.

mfocchi, 2020-11-11:
@jcarpent  
ok this was the issue, still a bit unclear to me what's the difference between and robotpkg-eigenpy and robotpkg-py27-eigenpy, the first one goes only till version 1.4.0r2

jcarpent, 2020-11-11:
Nice. 
Starting from 16.04, we are supporting both Python 2.7 and 3.5. So, we need to make the distinction between both, with either `robotpkg-py27-eigenpy` or `robotpkg-py35-eigenpy`.  `robotpkg-eigenpy` was formerly used for Python 2.7. We need to change this convention when starting to officially surpoting Python 3.x.

jcarpent, 2020-11-12:
I consider this issue to be solved. Michele, if you need further information, do not hesitate to ask more.
Best,

Justin

closed 2020-11-12 by jcarpent