stack-of-tasks/pinocchio #1340: Link Error related to URDF function

Richard31S, 2020-11-15:
Hi, 
I have just installed pinnochio and dependancies from sources. As beginner with pinocchio, I started on examples from https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/.
For first example, I have no issue and everything goes as described.
For the second one, I encounter one Link error related with URDF functions! I really don't know what's wrong! Here follows the Linker output:

/usr/bin/ld : /tmp/ccyRHFfY.o : dans la fonction « pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>& pinocchio::urdf::buildModel<double, 0, pinocchio::JointCollectionDefaultTpl>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>&, bool) » :
test.cpp:(.text._ZN9pinocchio4urdf10buildModelIdLi0ENS_25JointCollectionDefaultTplEEERNS_8ModelTplIT_XT0_ET1_EERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_b[_ZN9pinocchio4urdf10buildModelIdLi0ENS_25JointCollectionDefaultTplEEERNS_8ModelTplIT_XT0_ET1_EERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_b]+0x5b) : référence indéfinie vers « pinocchio::urdf::details::parseRootTree(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::urdf::details::UrdfVisitorBaseTpl<double, 0>&) »
collect2: error: ld returned 1 exit status

I use following build command line:
g++ -I `pkg-config --cflags --libs eigen3 pinocchio` -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_WITH_URDFDOM test.cpp -o overview-urdf

Here is the output of pkg-config --cflags --libs eigen3 pinocchio:
-DPINOCCHIO_WITH_URDFDOM -DBOOST_MPL_LIMIT_LIST_SIZE=30 -I/usr/local/include -I/usr/include/eigen3 -L/usr/local/lib -lpinocchio -lboost_filesystem -lboost_serialization -lboost_system -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -lconsole_bridge

Is it related to compatibility break between urdf libs and pinocchio versions ?
Thank for your help,
Richard

jcarpent, 2020-11-15:
Hi @Richard31S,

You just need to compile with C++11 support, because URDFDOM since version 1.0.0 is compiled against C++11.
Best,

Justin

Richard31S, 2020-11-15:
Hi @jcarpent,

Thank u for your answer. I am not familiar with such std support so I have just tried to recompile same file with the following command line:
g++ **-std=c++11** `pkg-config --cflags --libs pinocchio eigen3` -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_WITH_URDFDOM test.cpp -o overview-urdf

Unfortunately, I still have the **same** linker error... Any other suggestion?
Thank you,

Richard

jcarpent, 2020-11-16:
On which OS are you working on? Could you try with `-std=c++0x`? Could you provide your example?

Richard31S, 2020-11-16:
OS: Unbuntu 20.04 LTS.
I tried with -std=c++0x but nothing has changed.
My example is the one given at https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/:

#include "pinocchio/parsers/urdf.hpp"
#include "pinocchio/algorithm/joint-configuration.hpp"
#include "pinocchio/algorithm/kinematics.hpp"
#include <iostream>
// PINOCCHIO_MODEL_DIR is defined by the CMake but you can define your own directory here.
#ifndef PINOCCHIO_MODEL_DIR
  #define PINOCCHIO_MODEL_DIR "path_to_the_model_dir"
#endif
int main(int argc, char ** argv)
{
  using namespace pinocchio;
  
  // You should change here to set up your own URDF file or just pass it as an argument of this example.
  const std::string urdf_filename = (argc<=1) ? PINOCCHIO_MODEL_DIR + std::string("/others/robots/ur_description/urdf/ur5_robot.urdf") : argv[1];
  
  // Load the urdf model
  Model model;
  pinocchio::urdf::buildModel(urdf_filename,model);
  std::cout << "model name: " << model.name << std::endl;
  
  // Create data required by the algorithms
  Data data(model);
  
  // Sample a random configuration
  Eigen::VectorXd q = randomConfiguration(model);
  std::cout << "q: " << q.transpose() << std::endl;
  // Perform the forward kinematics over the kinematic tree
  forwardKinematics(model,data,q);
  // Print out the placement of each joint of the kinematic tree
  for(JointIndex joint_id = 0; joint_id < (JointIndex)model.njoints; ++joint_id)
    std::cout << std::setw(24) << std::left
              << model.names[joint_id] << ": "
              << std::fixed << std::setprecision(2)
              << data.oMi[joint_id].translation().transpose()
              << std::endl;
}

I installed Pinocchio from sources and I follow the tuto from  https://stack-of-tasks.github.io/pinocchio/download.html "Build from Source". The only thing I changed is the cmake call procedure:I called cmake-gui to remove Python Binding and Test features.

Thank u,

jcarpent, 2020-11-16:
```bash
g++ -std=c++11 $(pkg-config --cflags --libs pinocchio) overview-urdf.cpp -o test
``` 
works perfectly on my side

jcarpent, 2020-11-16:
Your error is coming from the fact you are not linking to the library itself

Richard31S, 2020-11-16:
Ok, all of this is not obvious for me so I try to explain my confusion:
In this example, if I comment "pinocchio::urdf::buildModel(urdf_filename,model);" line, link error disappears and everything is ok. So i guess, my pinocchio.so is consistent and well linked on this used perimeter. This has been demonstrated by previous example for which execution is ok too.
Now, I understand that the issue may be because of the urdfdom dependancie that is not well "done" and only required if I use URDF? Maybe, this library is not consistent. I install it in standalone from sources from https://github.com/ros/urdfdom.git as I don't want complete ROS.
Do you have any suggestion for me to move forward with URDF support ?
Thanks,

jcarpent, 2020-11-16:
Did you try:

> ```shell
> g++ -std=c++11 $(pkg-config --cflags --libs pinocchio) overview-urdf.cpp -o test
> ```
> 
> works perfectly on my side

Richard31S, 2020-11-16:
Yes, but I get same result:
g++ -std=c++11 $(pkg-config --cflags --libs pinocchio) overview-urdf.cpp -o test
/usr/bin/ld : /tmp/ccXuQqkL.o : dans la fonction « pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>& pinocchio::urdf::buildModel<double, 0, pinocchio::JointCollectionDefaultTpl>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>&, bool) » :
overview-urdf.cpp:(.text._ZN9pinocchio4urdf10buildModelIdLi0ENS_25JointCollectionDefaultTplEEERNS_8ModelTplIT_XT0_ET1_EERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_b[_ZN9pinocchio4urdf10buildModelIdLi0ENS_25JointCollectionDefaultTplEEERNS_8ModelTplIT_XT0_ET1_EERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_b]+0x5b) : référence indéfinie vers « pinocchio::urdf::details::parseRootTree(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pinocchio::urdf::details::UrdfVisitorBaseTpl<double, 0>&) »
collect2: error: ld returned 1 exit status

With: pkg-config --cflags --libs pinocchio
-DPINOCCHIO_WITH_URDFDOM -DBOOST_MPL_LIMIT_LIST_SIZE=30 -I/usr/local/include -I/usr/include/eigen3 -L/usr/local/lib -lpinocchio -lboost_filesystem -lboost_serialization -lboost_system -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -lconsole_bridge

I am really worried about this issue because pinocchio lib would fit perfectly my needs and my plans were to use it and benchmark on my Raspberry pi...

jcarpent, 2020-11-16:
You should define your LD_LIBRARY_PATH in such a way it points to the installationlocation of Pinocchio and Urdfdom. Where did you install the libs?

Richard31S, 2020-11-16:
I installed libs in /usr/local/lib and here follows the extract of .so files of this directory:
lrwxrwxrwx  1 root root      21 nov.  16 08:55 libpinocchio.so -> libpinocchio.so.2.5.3
-rw-r--r--  1 root root  265960 nov.  16 09:26 libpinocchio.so.2.5.3
lrwxrwxrwx  1 root root      23 nov.  16 00:03 liburdfdom_model.so -> liburdfdom_model.so.1.0
-rw-r--r--  1 root root  194744 nov.  15 23:54 liburdfdom_model.so.1.0
lrwxrwxrwx  1 root root      29 nov.  16 00:03 liburdfdom_model_state.so -> liburdfdom_model_state.so.1.0
-rw-r--r--  1 root root   57136 nov.  15 23:55 liburdfdom_model_state.so.1.0
lrwxrwxrwx  1 root root      24 nov.  16 00:03 liburdfdom_sensor.so -> liburdfdom_sensor.so.1.0
-rw-r--r--  1 root root   48248 nov.  15 23:54 liburdfdom_sensor.so.1.0
lrwxrwxrwx  1 root root      23 nov.  16 00:03 liburdfdom_world.so -> liburdfdom_world.so.1.0
-rw-r--r--  1 root root  195064 nov.  15 23:54 liburdfdom_world.so.1.0

LD_LIBRARY_PATH var is correctly set to this directory. I have already checked these points.
My issue is quite similar to "build error after update pinocchio #1196".

Can I compile pinocchio library with -static to "force" consistent build of it with URDFDOM dependancies?
Not sure this will help.

jcarpent, 2020-11-16:
Did you compile Pinocchio with C++11 too?

jcarpent, 2020-11-16:
Can you also show the output of:
```bash
nm /usr/local/lib/libpinocchio.dylib | grep  parseRootTree
```

Richard31S, 2020-11-16:
Yes, I checked that -std=gnu++11 is in command lines generated by make -j4...
Here follows some outputs:
nm /usr/local/lib/libpinocchio.so | grep  parseRootTree
0000000000013dc0 T _ZN9pinocchio4urdf7details13parseRootTreeEPKN4urdf14ModelInterfaceERNS1_18UrdfVisitorBaseTplIdLi0EEE
0000000000010996 t _ZN9pinocchio4urdf7details13parseRootTreeEPKN4urdf14ModelInterfaceERNS1_18UrdfVisitorBaseTplIdLi0EEE.cold
00000000000141e0 T _ZN9pinocchio4urdf7details13parseRootTreeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS1_18UrdfVisitorBaseTplIdLi0EEE
00000000000109ca t _ZN9pinocchio4urdf7details13parseRootTreeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS1_18UrdfVisitorBaseTplIdLi0EEE.cold
0000000000014300 T _ZN9pinocchio4urdf7details20parseRootTreeFromXMLERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS1_18UrdfVisitorBaseTplIdLi0EEE
0000000000010ab7 t _ZN9pinocchio4urdf7details20parseRootTreeFromXMLERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS1_18UrdfVisitorBaseTplIdLi0EEE.cold

ldd /usr/local/lib/libpinocchio.so
	linux-vdso.so.1 (0x00007ffe50317000)
	libboost_filesystem.so.1.71.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007f266da2c000)
	liburdfdom_model.so.1.0 => /usr/local/lib/liburdfdom_model.so.1.0 (0x00007f266d9ef000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f266d80e000)
	libgcc_s.so.1 => /usr/lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f266d7f3000)
	libc.so.6 => /usr/lib/x86_64-linux-gnu/libc.so.6 (0x00007f266d601000)
	libpthread.so.0 => /usr/lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f266d5de000)
	libtinyxml.so.2.6.2 => /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007f266d5c4000)
	libconsole_bridge.so.0.4 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007f266d5be000)
	libm.so.6 => /usr/lib/x86_64-linux-gnu/libm.so.6 (0x00007f266d46f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f266da81000)

nim65s, 2020-11-16:
Hi @Richard31S,

Could you try with `g++ overview-urdf.cpp $(pkg-config --cflags --libs pinocchio)` instead ?

(thx @nightmared :D)

Richard31S, 2020-11-16:
Hi @nim65s,

Yes !!!! It works perfectly now!!! Thank you very much!
So source file shall come first in the command line! Before closing this issue, can you explain what happens in linking process otherwise?
It would be nice to update the g++ command line given in tutorial...

Thank u so much Guys!
Richard

jcarpent, 2020-11-16:
> It would be nice to update the g++ command line given in tutorial...

Done in #1342.

nim65s, 2020-11-16:
@Richard31S here is an explanation on the importance of order at the link step: https://stackoverflow.com/a/409470/1368502

jcarpent, 2020-11-16:
Thanks @nim65s for concluding this issue. Hope @Richard31S you feel satisfied.

closed 2020-11-16 by jcarpent

edward9503, 2021-09-27:
Hi there,

I've got the same problem when I want to build the model from URDF. I installed Pinocchio and I follow the tuto from https://stack-of-tasks.github.io/pinocchio/download.html in section "Linux". All the libraries were installed in /opt/openrobots/lib. Then I tried to use these libraries to build the model from URDF file in my catkin ros workspace. When I run catkin_build, I got the same link error as aforementioned. Moerover, I saw some previous issues that I need to add the CMAKE_CXX_FLAGS in CmakeList.txt. But this still didnt work. So I was wondering where is the problem?

Here is my CmakeList.txt:

_> cmake_minimum_required(VERSION 2.8.3)
> project(unitree_controller)
> 
> ## Set the path to include model headers. Needs to be modified after using different model.
> set(MODEL_HEADER /usr/local/include/UnitreeA1/rcg CACHE PATH "Path to robot model headers")
> 
> ## Set the path to Pinocchio.
> set(PINOCCHIO_HEADER /opt/openrobots/include CACHE PATH "Path to Pinocchio")
> 
> ## Set the sources (C++) of your robot models. 
> set(SOURCES ../model/transforms.cpp
>             ../model/jacobians.cpp
>             ../model/jsim.cpp
>             ../model/inverse_dynamics.cpp
>             ../model/forward_dynamics.cpp
>             ../model/inertia_properties.cpp
>             ../model/miscellaneous.cpp)
> 
> find_package(catkin REQUIRED COMPONENTS
>     controller_manager
>     genmsg
>     joint_state_controller
>     robot_state_publisher
>     roscpp
>     gazebo_ros
>     std_msgs
>     tf
>     geometry_msgs
>     unitree_legged_msgs
>     rosbag
>     urdf
>     kdl_parser
> )
> 
> find_package(PkgConfig REQUIRED)
> pkg_check_modules(PINOCCHIO pinocchio REQUIRED)
> 
> find_package(gazebo REQUIRED)
> 
> catkin_package(
>     CATKIN_DEPENDS 
>     unitree_legged_msgs 
> )
> 
> include_directories(
>     include
>     ${Boost_INCLUDE_DIR}
>     ${catkin_INCLUDE_DIRS}
>     ${GAZEBO_INCLUDE_DIRS}
>     ${MODEL_HEADER}
>     ${PINOCCHIO_HEADER}
> )
> 
> link_directories(${GAZEBO_LIBRARY_DIRS} /opt/openrobots/lib)
> set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${GAZEBO_CXX_FLAGS}")
> 
> 
> # Declare a C++ library
> add_library(${PROJECT_NAME}
>     src/body.cpp 
>     src/a1body.cpp
>     ${SOURCES}
> )
> 
> add_dependencies(${PROJECT_NAME} unitree_legged_msgs_gencpp)
> 
> target_link_libraries(${PROJECT_NAME}
>   ${catkin_LIBRARIES} ${EXTRA_LIBS}
> )
> 
> # add_library(unitreeFootContactPlugin SHARED plugin/foot_contact_plugin.cc)
> # target_link_libraries(unitreeFootContactPlugin ${catkin_LIBRARIES} ${GAZEBO_LIBRARIES})
> 
> # add_library(unitreeDrawForcePlugin SHARED plugin/draw_force_plugin.cc)
> # target_link_libraries(unitreeDrawForcePlugin ${catkin_LIBRARIES} ${GAZEBO_LIBRARIES})
> 
> add_executable(unitree_external_force src/external_force.cpp)
> target_link_libraries(unitree_external_force ${catkin_LIBRARIES})
> 
> add_executable(unitree_servo src/servo.cpp)
> target_link_libraries(unitree_servo ${PROJECT_NAME} ${catkin_LIBRARIES})
> 
> add_executable(position_mode src/position_exe.cpp)
> target_link_libraries(position_mode ${PROJECT_NAME} ${catkin_LIBRARIES})
> 
> add_executable(model_test_FK src/model_test_FK.cpp)
> target_link_libraries(model_test_FK ${PROJECT_NAME} ${catkin_LIBRARIES})
> 
> add_executable(RobCoGen_VS_Pinocchio src/RobCoGen_VS_Pinocchio.cpp)
> target_link_libraries(RobCoGen_VS_Pinocchio ${PROJECT_NAME} ${catkin_LIBRARIES})
> target_compile_options(RobCoGen_VS_Pinocchio PUBLIC ${PINOCCHIO_CFLAGS_OTHER})
> 
> add_executable(velocity_mode src/velocity_exe.cpp)
> target_link_libraries(velocity_mode ${PROJECT_NAME} ${catkin_LIBRARIES})
> 
> add_executable(torque_mode src/torque_exe.cpp)
> target_link_libraries(torque_mode ${PROJECT_NAME} ${catkin_LIBRARIES})
> 
> add_executable(unitree_move_kinetic src/move_publisher.cpp)
> target_link_libraries(unitree_move_kinetic ${catkin_LIBRARIES})_

When I typed "pkg-config --cflags --libs pinocchio", I got following output:

> _-DPINOCCHIO_WITH_URDFDOM -DPINOCCHIO_WITH_HPP_FCL -DHPP_FCL_HAS_OCTOMAP -DHPP_FCL_HAVE_OCTOMAP -DFCL_HAVE_OCTOMAP -DOCTOMAP_MAJOR_VERSION=1 -DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=6 -I/opt/openrobots/include -I/usr/local/include/eigen3 -L/opt/openrobots/lib -Wl,-rpath,/opt/openrobots/lib -lpinocchio -Wl,-rpath,/usr/lib/x86_64-linux-gnu -lboost_filesystem -lboost_serialization -lboost_system -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -lconsole_bridge -Wl,-rpath,/opt/openrobots/lib -lhpp-fcl -loctomap -loctomath_

Many thanks in advance!