stack-of-tasks/pinocchio #1343: Inverse Dynamics of Floating Base system

aabouman, 2020-11-19:
The function `pinocchio.rnea(model, data, q, v, a)` returns a vector of the input torques to achieve the desired acceleration given the current state. The size of this vector is equal to the size of the state of the system, this output is only valid for a fully actuated system (# states = # control inputs). Is it possible to use pinocchio to solve an inverse dynamics problem for a system which has fewer inputs than states (for example a legged robot)? Any advice would be greatly appreciated!

jcarpent, 2020-11-19:
You should use https://github.com/stack-of-tasks/tsid for solving such a problem. TSID (aka Task Space Inverse Dynamics) implements the request features and comes with Python bindings. It is of course based on Pinocchio ;)

jcarpent, 2020-11-19:
This issue seems to be solved then.

closed 2020-11-19 by jcarpent