stack-of-tasks/pinocchio #1345: Pinocchio incompatible with Eigen 3.3.9

Synxis, 2020-11-20:
Pinocchio does not compile against Eigen >= 3.3.9, due to the file `pinocchio/utils/eigen-fix.hpp`. Here is the output when compiling with clang 10 on Ubuntu 18.04:

```
In file included from $PROJECT_DIR/third-parties/eigen/eigen/Eigen/LU:29:0,
                 from $PROJECT_DIR/third-parties/eigen/eigen/Eigen/Geometry:14,
                 from $PROJECT_DIR/third-parties/pinocchio/include/pinocchio/math/quaternion.hpp:19,
                 from $PROJECT_DIR/third-parties/pinocchio/include/pinocchio/spatial/se3-tpl.hpp:12,
                 from $PROJECT_DIR/third-parties/pinocchio/include/pinocchio/spatial/se3.hpp:44,
                 from $PROJECT_DIR/third-parties/pinocchio/include/pinocchio/multibody/model.hpp:10,
                 from $PROJECT_DIR/third-parties/pinocchio/include/pinocchio/parsers/urdf.hpp:9,
                 from $PROJECT_DIR/third-parties/pinocchio/pinocchio/src/parsers/urdf/geometry.cpp:5:
$PROJECT_DIR/third-parties/eigen/eigen/Eigen/src/LU/PartialPivLU.h: In static member function ‘static Eigen::Index Eigen::internal::partial_lu_impl<Scalar, StorageOrder, PivIndex, SizeAtCompileTime>::unblocked_lu(Eigen::internal::partial_lu_impl<Scalar, StorageOrder, PivIndex, SizeAtCompileTime>::MatrixTypeRef&, PivIndex*, PivIndex&)’:
$PROJECT_DIR/third-parties/eigen/eigen/Eigen/src/LU/PartialPivLU.h:390:27: error: expected primary-expression before ‘<’ token
         lu.col(k).tail(fix<RRows>(rrows)) /= lu.coeff(k,k);
                           ^
$PROJECT_DIR/third-parties/eigen/eigen/Eigen/src/LU/PartialPivLU.h:400:33: error: expected primary-expression before ‘<’ token
         lu.bottomRightCorner(fix<RRows>(rrows),fix<RCols>(rcols)).noalias() -= lu.col(k).tail(fix<RRows>(rrows)) * lu.row(k).tail(fix<RCols>(rcols));
                                 ^
$PROJECT_DIR/third-parties/eigen/eigen/Eigen/src/LU/PartialPivLU.h:400:51: error: expected primary-expression before ‘<’ token
         lu.bottomRightCorner(fix<RRows>(rrows),fix<RCols>(rcols)).noalias() -= lu.col(k).tail(fix<RRows>(rrows)) * lu.row(k).tail(fix<RCols>(rcols));
                                                   ^
$PROJECT_DIR/third-parties/eigen/eigen/Eigen/src/LU/PartialPivLU.h:400:98: error: expected primary-expression before ‘<’ token
    lu.bottomRightCorner(fix<RRows>(rrows),fix<RCols>(rcols)).noalias() -= lu.col(k).tail(fix<RRows>(rrows)) * lu.row(k).tail(fix<RCols>(rcols));
                                                                                             ^
$PROJECT_DIR/third-parties/eigen/eigen/Eigen/src/LU/PartialPivLU.h:400:134: error: expected primary-expression before ‘<’ token
 rows),fix<RCols>(rcols)).noalias() -= lu.col(k).tail(fix<RRows>(rrows)) * lu.row(k).tail(fix<RCols>(rcols));
                                                                                             ^
```

1. Is this code necessary ?
2. Why put it in `Eigen::internal` ? This is brittle as it depends on stuff that can change without notice (case in point...), it would be better to put it in some namespace `eigenfix` (for example)

jcarpent, 2020-11-20:
Thanks for this raising this issue.

It seems that you are bit in advance @Synxis as Eigen 3.3.9 is not released yet. We cannot on our side try to remain compatible with all the dependencies (on many platforms) + trying to make Pinocchio compatible with future versions of dependencies that are not yet released.

Concerning the point 1., the answer is yes.
For the second point, you're right. I could not imagine when I wrote this code that Eigen would add something called fix.
I changed the rule in the new PR #1346. Your review is really welcome.

Synxis, 2020-11-20:
Thank you for your very quick response (probably the fastest I've ever had on Github) !

I confirm that it now compiles correctly on my machine with Eigen 3.3.9.

I agree that supporting all versions of all dependencies is a very high burden, but not polluting the namespaces of dependencies help a lot. Your fix is the right way to do it, thanks !

jcarpent, 2020-11-20:
Solved by #1346.

closed 2020-11-20 by jcarpent