stack-of-tasks/pinocchio #1352: Memory allocation segfault when compiling in Debug

duburcqa, 2020-12-02:
Hi,

In order to debug a segfault on my side, I compiled Pinocchio in Debug, and I noticed a segmentation fault when creating a `Pinocchio::Model` without argument. Surprisingly, I don't have such issue in Release. I can try to provide a snippet to reproduce the problem if necessary. Here is the gdb backtrace:

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff616e921 in __GI_abort () at abort.c:79
#2  0x00007ffff61b7967 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff62e4b0d "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff61be9da in malloc_printerr (str=str@entry=0x7ffff62e2d8e "malloc(): memory corruption") at malloc.c:5342
#4  0x00007ffff61c2b24 in _int_malloc (av=av@entry=0x7ffff6519c40 <main_arena>, bytes=bytes@entry=80) at malloc.c:3748
#5  0x00007ffff61c51cc in __GI___libc_malloc (bytes=80) at malloc.c:3067
#6  0x00005555555aeb52 in Eigen::internal::aligned_malloc (size=80) at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:159
#7  0x000055555561e151 in Eigen::aligned_allocator<pinocchio::InertiaTpl<double, 0> >::allocate (this=0x5555564b5518, num=1) at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:742
#8  0x0000555555613eaa in std::allocator_traits<Eigen::aligned_allocator<pinocchio::InertiaTpl<double, 0> > >::allocate (__a=..., __n=1) at /usr/include/c++/7/bits/alloc_traits.h:301
#9  0x000055555560a2be in std::_Vector_base<pinocchio::InertiaTpl<double, 0>, Eigen::aligned_allocator<pinocchio::InertiaTpl<double, 0> > >::_M_allocate (this=0x5555564b5518, __n=1)
    at /usr/include/c++/7/bits/stl_vector.h:172
#10 0x0000555555994611 in std::_Vector_base<pinocchio::InertiaTpl<double, 0>, Eigen::aligned_allocator<pinocchio::InertiaTpl<double, 0> > >::_M_create_storage (this=0x5555564b5518, __n=1)
    at /usr/include/c++/7/bits/stl_vector.h:187
#11 0x000055555598f591 in std::_Vector_base<pinocchio::InertiaTpl<double, 0>, Eigen::aligned_allocator<pinocchio::InertiaTpl<double, 0> > >::_Vector_base (this=0x5555564b5518, __n=1, __a=...)
    at /usr/include/c++/7/bits/stl_vector.h:138
#12 0x00005555559894de in std::vector<pinocchio::InertiaTpl<double, 0>, Eigen::aligned_allocator<pinocchio::InertiaTpl<double, 0> > >::vector (this=0x5555564b5518, __n=1, __value=..., __a=...)
    at /usr/include/c++/7/bits/stl_vector.h:297
#13 0x00005555559818db in pinocchio::container::aligned_vector<pinocchio::InertiaTpl<double, 0> >::aligned_vector (this=0x5555564b5518, num=1, val=...)
    at /opt/openrobots/include/pinocchio/container/aligned-vector.hpp:43
#14 0x0000555555979982 in pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>::ModelTpl (this=0x5555564b5500) at /opt/openrobots/include/pinocchio/multibody/model.hpp:177
#15 0x000055555596d705 in jiminy::Model::Model (this=0x5555564b4510) at /home/alexis/wdc_workspace/src/jiminy/core/src/robot/Model.cc:54
#16 0x0000555555687ba4 in jiminy::Robot::Robot (this=0x5555564b4510) at /home/alexis/wdc_workspace/src/jiminy/core/src/robot/Robot.cc:40
#17 0x00005555555bef7f in __gnu_cxx::new_allocator<jiminy::Robot>::construct<jiminy::Robot> (this=0x7fffffffd237, __p=0x5555564b4510) at /usr/include/c++/7/ext/new_allocator.h:136
#18 0x00005555555bd2af in std::allocator_traits<std::allocator<jiminy::Robot> >::construct<jiminy::Robot> (__a=..., __p=0x5555564b4510) at /usr/include/c++/7/bits/alloc_traits.h:475
#19 0x00005555555bab08 in std::_Sp_counted_ptr_inplace<jiminy::Robot, std::allocator<jiminy::Robot>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<>(std::allocator<jiminy::Robot>) (
    this=0x5555564b4500, __a=...) at /usr/include/c++/7/bits/shared_ptr_base.h:526
#20 0x00005555555b8516 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<jiminy::Robot, std::allocator<jiminy::Robot>>(std::_Sp_make_shared_tag, jiminy::Robot*, std::allocator<jiminy::Robot> const&) (this=0x7fffffffd3a8, __a=...) at /usr/include/c++/7/bits/shared_ptr_base.h:637
#21 0x00005555555b6d22 in std::__shared_ptr<jiminy::Robot, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<jiminy::Robot>>(std::_Sp_make_shared_tag, std::allocator<jiminy::Robot> const&) (
    this=0x7fffffffd3a0, __tag=..., __a=...) at /usr/include/c++/7/bits/shared_ptr_base.h:1295
#22 0x00005555555b4a12 in std::shared_ptr<jiminy::Robot>::shared_ptr<std::allocator<jiminy::Robot>>(std::_Sp_make_shared_tag, std::allocator<jiminy::Robot> const&) (this=0x7fffffffd3a0, __tag=..., 
    __a=...) at /usr/include/c++/7/bits/shared_ptr.h:344
#23 0x00005555555b2963 in std::allocate_shared<jiminy::Robot, std::allocator<jiminy::Robot>>(std::allocator<jiminy::Robot> const&) (__a=...) at /usr/include/c++/7/bits/shared_ptr.h:691
#24 0x00005555555b0f55 in std::make_shared<jiminy::Robot> () at /usr/include/c++/7/bits/shared_ptr.h:707
#25 0x00005555555ab73b in main (argc=1, argv=0x7fffffffd718) at /home/alexis/wdc_workspace/src/jiminy/examples/double_pendulum/double_pendulum.cc:73
```

jmirabel, 2020-12-02:
This looks like a memory alignment issue. Are the compilation options the same for all the binaries used in your testcase ?

duburcqa, 2020-12-02:
Hum... I dont know. Probably not because I didn't do anything about it. Unfortunately, I'm not aware of how to do it. Can you point me out where it is mentioned in the documentation how to deal with memory alignment ? Sorry for the inconvenience :/

jmirabel, 2020-12-02:
Debug and release will not cause a bug as above. What can cause such a bug is turning on options such as SSE operations for some libraries only.

If possible for you, compile all libraries in debug mode. You may have some useful Eigen assertion popping up.

cmastalli, 2020-12-02:
> Can you point me out where it is mentioned in the documentation how to deal with memory alignment ? Sorry for the inconvenience :/

Pinocchio support memory alignments. You can enable by defining the following compilation flag `-march=native`. For learning how your code (derived from Pinocchio) could support memory alignment, you should revise the Eigen documentation.

duburcqa, 2020-12-02:
OK thank you for your insights. Since it seems that it is not an issue with Pinocchio itself I'm closing this issue.

closed 2020-12-02 by duburcqa

jcarpent, 2020-12-02:
Please look at the unit tests: https://github.com/stack-of-tasks/pinocchio/blob/29be057af1beb47f2ad8f411e1c1fb7acced74f7/unittest/model.cpp#L373-L383
We are already checking the empty model construction.