stack-of-tasks/pinocchio #1355: Installation Issues

Neotriple, 2020-12-09:
Hi,

I'm trying to install pinocchio using the following method:

https://stack-of-tasks.github.io/pinocchio/download.html

However, when I run the command 

`sudo apt install -qqy robotpkg-py36-pinocchio`

I receive the following error:

```
The following packages have unmet dependencies:
 robotpkg-py36-pinocchio : Depends: robotpkg-pinocchio (= 2.5.3) but 2.5.4 is to be installed
E: Unable to correct problems, you have held broken packages.
```

I tried to install robotpkg-pinocchio and then robotpkg-py36-pinocchio but ran into the same error. I also ran into the same error when attempting to install the py-27 version.

jcarpent, 2020-12-09:
I think @nim65s is the right person to help you.

nim65s, 2020-12-10:
Hi @Neotriple,

I guess you updated your package index while the packages where being built. Can you try again after another `sudo apt update` ?

jcarpent, 2020-12-13:
I will close this issue. Thanks @nim65s for answering.

closed 2020-12-13 by jcarpent