stack-of-tasks/pinocchio #1356: Casadi-Pinocchio interface

andreadelprete, 2020-12-10:
Hi, together with @gfadini @thomasfla @psoueres we're investigating the use of Casadi for optimal control. However, we'd like not to give up Pinocchio for the dynamics. I see that Pinocchio is somewhat interfaced with Casadi, but I cannot figure out if this pinocchio-casadi interface is complete (at least for c++). I've seen there was another issue open about a similar subject #911 but that issue seemed to focus on the python code, so I though to open a new one.

Thanks in advance for the help!
Andrea

jcarpent, 2020-12-10:
Pinocchio is 100% compatible with Casadi and @alejandroastudillo can confirm it. 
The current CI is indeed checking this compatibility. Not all algorithms have been tested, but if you encounter troubles, I might help you to fix it. It is often due to Eigen and not Pinocchio.

For the Python bindings, if it simplifies your life, it might be worth to finish it.
I've implemented the interface between Casadi/Eigen/Numpy via EigenPy. The only missing stuff is to make the Python bindings generic in the sense we can switch from double to any Scalar types.

cmastalli, 2020-12-10:
> I've implemented the interface between Casadi/Eigen/Numpy via EigenPy. The only missing stuff is to make the Python bindings generic in the sense we can switch from double to any Scalar types.

How does this work affect the integration of CppAD? It would be wonderful!

jcarpent, 2020-12-10:
You need to be able to share the memory between Numpy and Eigen, and this is done by EigenPy ;)

cmastalli, 2020-12-10:
> We can do similar things with CppAD and CppADCodeGen too.

I would like to help you with this task, but I somehow miss the needed understanding of EigenPy.
What I could offer you if to be a beta tester :)

This task will put definitely Pinocchio in another level :+1:

thomasfla, 2020-12-18:
Thanks @jcarpent, having the python bindings would be really nice! Can you give us some hints on what needs to be done to implement them? 

Also, we came across this interface https://github.com/ADVRHumanoids/casadi_kin_dyn witch seems very simple to use in python. What do you think ?

jcarpent, 2020-12-18:
The remaining stuff is to make the current Python C++ module of Pinocchio, which exposes Pinocchio as Python module, independent from the Scalar type. This is not hard, but must be done carefully to be generic enough to allow exposition of other modules like CppAd and CppAdCodeGen.

The reference you mentioned won't be generic enough for your purpose I guess, and I would rather recommend working together on the first point, which increases the usage of Pinocchio.

cmastalli, 2020-12-18:
That code is binding a few Pinocchio functions that uses the CasADi scalar in c++.
You ideally want to do this operation at the Python level; otherwise, the code complexity still remains (i.e., writing things in c++).

gfadini, 2020-12-18:
Hi, being able to explore the features of casadi in cooperation with pincchio would be quite interesting to me and I guess to other users as well. Currently this is just about an explorative endeavour, I'd be glad to help on this side and be further instructed as @jcarpent suggested.
@cmastalli surely working at python ideally would be simpler, maybe you have a specific suggestion on this side?

proyan, 2021-08-24:
I'll close this issue, since this is already handled in pinocchio3-preview branch

closed 2021-08-24 by proyan

Ipuch, 2023-07-07:
Hi, 

I post my question here as the title seems relevant. 

I'm one of the core developers of [bioptim](github.com/pyomeca/bioptim).
Our framework is kind of a high-level interface to define optimal control problems for biomechanics through CasADi.

We recently disconnected [bioptim](github.com/pyomeca/biorbd) (relying on rbdl), our main software handling the multibody dynamics and musculoskeletal dynamics, from [bioptim](github.com/pyomeca/bioptim) to offer the capability to connect your own model and any software compatible with CasADi.

From the current issue, I read Pinocchio is compatible with CasADI and available in pinocchio3-preview. Can I install it from pip or conda, and use it through Python? 

Pierre Puchaud

stephane-caron, 2023-07-19:
You can get a preview of Pinocchio 3 from the [JNRH'2023: Pinocchio Tutorial](https://github.com/nmansard/jnrh2023), which ships a Conda binary release:

- Conda install: https://github.com/nmansard/jnrh2023#conda
- Support chat: https://matrix.to/#/#jnrh2023-tuto:laas.fr
- Slides: https://gepettoweb.laas.fr/talks/jnrh2023/

All tutorial notebooks use Pinocchio 3 + CasADi, for instance the one on [Trajectory optimization](https://github.com/nmansard/jnrh2023/blob/main/2_trajopt_geom.ipynb).

This is a binary preview in the meantime before the first official release of Pinocchio 3 @jcarpent.

jcarpent, 2023-07-25:
Dear Pierre (@lpuch),

Could you please email me to discuss your needs and see how much we can collaborate on musculoskeletal modeling? We are indeed planning to add this support in native within Pinocchio, and it might be the occasion to think of a common interface.

Best,

Justin