stack-of-tasks/pinocchio #1357: Generalized velocity of the floating base

mayataka, 2020-12-13:
Hi all,

I want to know which reference frame the generalized velocity of the floating base is actually represented. 
In #1122, the translation velocity of the floating base seems expressed in the WORLD frame.
However, in #1137, it seems expressed in the LOCAL frame (and the angular velocity seems also in the LOCAL frame), and I'm confused.
This point is important to simulate a controller based on Pinocchio on the other rigid-body simulators.
I'd appreciate it if you tell me this point.

jcarpent, 2020-12-13:
The placement variable in Pinocchio is following this convention: (t,R) \in SE(3), with t being the translation expressed in the WORLD frame (antecedent frame)
For the generalized velocity, we follow the convention of Featherstone to express it in the local Frame of the joint.
To get the expression in the World coordinate system centered on the base frame, you just need to rorate the Linear and Angular vectors using the R rotation matrix.

mayataka, 2020-12-13:
Thank you very much!
I understood completely :)

closed 2020-12-13 by mayataka