stack-of-tasks/pinocchio #1360: How to use Pinocchio to calculate G in this kinetic formula?

wjingdan, 2020-12-15:
Hello, I want to know how to use Pinocchio to calculate G in this kinetic formula? Thank you so much.

[image]

jcarpent, 2020-12-15:
What is the G matrix? It is related to some contact constraints? Could you send a reference to understand its meaning?

wjingdan, 2020-12-16:
[physcap-1.pdf](https://github.com/stack-of-tasks/pinocchio/files/5699796/physcap-1.pdf)

Hello. In the above paper, G is defined as such.

[image]

jcarpent, 2020-12-16:
OK. This quantity corresponds to the action of the contact transformation (placement quantity of the contact point w.r.t. to the supporting joint) on the contact forces. It is nothing more than:
```python
M = pin.SE3.Random() # placement of the contact point w.r.t. the joint frame
f = pin.Force(fext,np.zeros((3))) # value of the contact force expressed as a spatial Force quantity
f_joint = M.act(f)
```

jcarpent, 2020-12-16:
By the way, I don't think the phrasing of your reference paper is correct. There is a lot of ambiguities in the notation and the naming. If you need a correct reference, Featherstone is the correct book.

wjingdan, 2020-12-16:
Thank you for your answers. 
In the sample code: contact-dynamics.cpp, after this step (**pinocchio::forwardDynamics(model, data, q, v, tau, J, gamma, 0.);**), you directly call data.lambda_c like this (**Eigen ::VectorXd dynamics_residual (data.M * data.ddq + data.nle-tau-J.transpose()*data.lambda_c);**), so data.lambda_c is the contact force and moment of "rleg6_joint" and "lleg6_joint" ( dim=6)?
[image]

jcarpent, 2020-12-16:
Yes, exactly. But this is for rigid contact model.

wjingdan, 2020-12-16:
Thank you very much.

jcarpent, 2020-12-16:
Fine. I will close this issue then.

closed 2020-12-16 by jcarpent

BERSERKPH52, 2021-03-02:
> OK. This quantity corresponds to the action of the contact transformation (placement quantity of the contact point w.r.t. to the supporting joint) on the contact forces. It is nothing more than: [...]
Hi, jcarpent.
I'm a beginner of Pinocchio and I also encountered the same problem when calculating G.
After reading your reply, I still have a question about how **f_joint** can know which joint frame it should be with respect to, because I didn't see any input about joint information.  
I‘d appreciate it if you can answer.

jcarpent, 2021-03-02:
G is nothing more than the transformation that maps the contact point locomation to the joint local frame.
If you know the placement M_i of the contact point i, then you need to simply call `M_i.toDualActionMatrix().leftCols<3>()`.

jcarpent, 2021-03-02:
By the way, a more complete approach for solving similar problems has been done by @zongmianli: https://www.di.ens.fr/willow/research/motionforcesfromvideo/research/li19mfv/

The whole implementation based on Pinocchio will be made open source very soon.

BERSERKPH52, 2021-03-02:
Thank you very much for your patience to explain and share, I will follow zongmianli‘s work in the future.

luyangzhu, 2021-05-11:
Hi Justin, any rough timeline for releasing the implementation for zongmian's paper? I am also new to pinocchio, want to find some tutorials of pinocchio on the inverse dynamics control with contact constraints? (e.g., a tutorial on how to control humanoid robot to track a reference walking motion considering the foot contacts)