stack-of-tasks/pinocchio #1362: Compute Robot's Inertia as reflected in the base frame

mrsp, 2020-12-17:
Hi I want to compute the Angular momentum around the CoM in the world frame namely,

L = R_b * ( I * omega_b + L_kin)

Lets assume I have the Rotation from the base to the world frame **R_b** and the local angular velocity of the base **omega_b** available. 

I think I can compute **L_kin** with the pinnochio **computeCentroidalMomentum** function (right?).

But is there a way to compute the Inertia of the robot as seen by the base frame **I** with pinnochio?

Thanks in advance

jcarpent, 2020-12-17:
The centroidal inertia quantity is computed by `computeCentroidalMap`.  The centroidal inertia expressed in the WORLD coordinate system (origin of the world), is then given by `data.oYcrb[0]`.

Assuming the base frame is related to the first joint_id, then the centroidal inertia expressed in the base frame is given by `data.oMi[1].actInv(data.oYcrb[0])`.

I hope this is clear.

mrsp, 2020-12-17:
Yes, thanks for the quick reply.

closed 2020-12-17 by mrsp

cmastalli, 2020-12-17:
Btw, there is no bindings for "data.oYcrb", only for "data.Ycrb as we can see here: https://github.com/stack-of-tasks/pinocchio/blob/17c479e2fe91905122713643f489feb8ddf135a3/bindings/python/multibody/data.hpp#L121

mrsp, 2020-12-17:
I am using the c++ implementation, if oYcrb is with respect to the origin Ycrb is with respect to ?

jcarpent, 2020-12-17:
With respect to the LOCAL joint frame.