stack-of-tasks/pinocchio #1367: Optionally build

nuft, 2021-01-07:
Hi, thanks for making pinocchio available!

We build pinocchio from source because we need CppAD and hpp-fcl self collision support.
Currently, a major issue for us is the build time of the optional bindings, unittests,  benchmarks, examples, etc.
Would it be possible to conditionally compile the optional parts with cmake `option(BUILD_WITH_ ....)`?

Cheers,
Michael

jcarpent, 2021-01-07:
hi @nuft,

First of all, thanks for using Pinocchio.

You can deactivate the build of the various settings by providing the following options to cmake:
- -DBUILD_TESTING=OFF
- -DBUILD_BENCHMARK=OFF
- -DBUILD_UTILS=OFF

Then, when you will do 'make install`, only the Python bindings and the main lib will be installed.
For using CppAD, you are not obliged to mention it to Pinocchio at compile time. Only your project should be linked to CppAD. In Pinocchio, we just use the dependency to CppAD for the unit testings.

nuft, 2021-01-07:
Hi @jcarpent thanks for the quick answer.

I wasn't aware that I could disable the examples with BUILD_TESTING flag.

We're looking for a solution to simplify the build steps and avoid having to passing the flags manually.
Also, we're using catkin as build system, so I'm trying to make the integration easier by using catkin_package() instead of PkgConfig. Probably in this case the way to go is a catkin wrapper package.

Thanks for your input

closed 2021-01-07 by nuft

jcarpent, 2021-01-07:
There is this archived repo: https://github.com/ipab-slmc/pinocchio_catkin
I think @wxmerkt may guide on this topic.

wxmerkt, 2021-01-07:
Hi @nuft,
If you are using this repository, you need to manually change the options flag defaults in the `CMakeLists.txt`. I am locally adding this change on every clone as there isn't a direct way (other than `catkin config --cmake-args -DBUILD_TESTING=OFF` for all packages).

If you are using `pinocchio_catkin`, I'd recommend to directly clone this repository instead (it will build with catkin as well).

nuft, 2021-01-11:
Hi @wxmerkt,
At the moment I'm changing the defaults in the CMakeLists.txt as well. For our CI server the changes need to be tracked by version control but I would like to avoid forking pinocchio just for that.
So the `pinocchio_catkin` wrapper package is probably the cleanest way to achieve this (I was hesitant about this since it was archived on GitHub)

wxmerkt, 2021-01-11:
Hi @nuft,
I am unaware of your particular CI setup but if you can add those parameters to the workspace settings it should work. I would strongly recommend against use of `pinocchio_catkin` - it's been archived for a while and we've not tested it since Pinocchio supports targets (and has changed the cmake module quite a bit). So either a fork with rebasing (and I agree it should be unnecessary for such a small change) or a workspace setting (in catkin or colcon) should do the trick.

nuft, 2021-01-11:
I see your concern but I will give it a try.
The issue with workspace settings is that it is not user friendly. it would add more steps to setting up a new workspace.
We want to keep the workflow simple: Just clone into the workspace and build.