stack-of-tasks/pinocchio #1370: Rendering issue when using scaling the mesh files in the URDF

skleff1994, 2021-01-12:
Hello,

As I was trying to display a KUKA robot in the gepetto viewer I noticed that scaling the mesh (.stl file) in the URDF causes a rendering problem. I isolated the issue in this minimal script for the KUKA arm and the double pendulum
```
import pinocchio as pin
from pinocchio import neutral
from pinocchio.robot_wrapper import RobotWrapper      

# urdf_filename = "/home/skleff/.local/lib/python3.6/site-packages/robot_properties_kuka/robot_properties_kuka/iiwa.urdf"                                                                                            
# mesh_dir = "/home/skleff/.local/lib/python3.6/site-packages/robot_properties_kuka"          

urdf_filename = "/home/skleff/double_pendulum_description/urdf/double_pendulum.urdf"                                                                                            
mesh_dir = "/home/skleff/double_pendulum_description"   

pin_robot = RobotWrapper.BuildFromURDF(urdf_filename, mesh_dir)                                                                                                                                                                                                                                                                                                                                                
q0 = neutral(pin_robot.model)                                                                                                                                                     
pin_robot.initDisplay(loadModel=True)
pin_robot.display(q0)
```

Here are the URDF files I am using . The KUKA URDF file was generated from a .xacro file (I edit it by hand here to exhibit the issue) . 

```
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from /home/skleff/.local/lib/python3.6/site-packages/robot_properties_kuka/robot_properties_kuka/xacro/iiwa.urdf.xacro | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="KUKAiiwa14" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <material name="Black">
    <color rgba="0.0 0.0 0.0 1.0"/>
  </material>
  <material name="Blue">
    <color rgba="0.0 0.0 0.8 1.0"/>
  </material>
  <material name="Green">
    <color rgba="0.0 0.8 0.0 1.0"/>
  </material>
  <material name="Grey">
    <color rgba="0.4 0.4 0.4 1.0"/>
  </material>
  <material name="Orange">
    <color rgba="1.0 0.4235294117647059 0.0392156862745098 1.0"/>
  </material>
  <material name="Brown">
    <color rgba="0.8705882352941177 0.8117647058823529 0.7647058823529411 1.0"/>
  </material>
  <material name="Red">
    <color rgba="0.8 0.0 0.0 1.0"/>
  </material>
  <material name="White">
    <color rgba="1.0 1.0 1.0 1.0"/>
  </material>
  <!-- Fix to world just for testing -->
  <link name="world"/>
  <!--joint between WORLD and BASE-->
  <joint name="A0" type="fixed">
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <parent link="world"/>
    <child link="iiwa_base"/>
  </joint>
  <!-- BASE -->
  <link name="iiwa_base">
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <mass value="5"/>
      <inertia ixx="0.05" ixy="0" ixz="0" iyy="0.06" iyz="0" izz="0.03"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
[... 539 lines cut ...]
      link="link2" />
    <axis
      xyz="1 0 0" />
    <limit
      lower="0"
      upper="0"
      effort="0"
      velocity="0" />
    <dynamics
      damping="0.05" />
  </joint>
</robot>
```

Here is the rendering in gepetto viewer when I use the `scale="..."` parameter for the mesh files : 

[image]

When I remove the scale parameter in the URDF I get a proper rendering - but obvioulsy not the right scale : 

[image]

I get the same outcome when I use a scaling parameter in the double pendulum URDF. I found [this post](https://answers.ros.org/question/340553/stl-model-disappears-if-scale-parameter-is-set-in-urdf-file/) about a similar issue, suggesting it could be a numeric encoding problem but I the proposed solution didn't solve my problem . 

Are you able to reproduce this issue with the script I provided ? If it's indeed a but, do you have any clue about a fix ?

jcarpent, 2021-01-12:
This is not a problem from Pinocchio at first glance. The first display is correct, you have the robot in the neutral configuration. But, I don't know why you don't have the texture. Did you try first with the other available viewers?
Could you directly ask people @ LAAS-CNRS?

PS: for next time, please make a script independent from your machine PATH.

jcarpent, 2021-01-12:
By the way, you can also try to convert the STL files into DAE, using `osgconv`

jmirabel, 2021-01-13:
Hi,

this is indeed a bug and it is known. When applying a large scale in OSG, sometimes you get this. I have no idea how to solve this is inside OSG directly.

I implemented a workaround in https://github.com/Gepetto/gepetto-viewer/pull/162.

jcarpent, 2021-01-13:
Thanks @jmirabel for the quick feedback. As it is not related to Pinocchio, I'm closing this issue.

closed 2021-01-13 by jcarpent

skleff1994, 2021-01-13:
Thank you for you prompt answer , I will use the workaround of @jmirabel