stack-of-tasks/pinocchio #1371: Problem using meshcat

jviereck, 2021-01-13:
I try to run the visualization demo for solo12 using meshcat [1]. For me this results in the following UserWarning:

```
/usr/local/Cellar/python@3.9/3.9.0_4/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py:127: RuntimeWarning: 'meshcat.servers.zmqserver' found in sys.modules after import of package 'meshcat.servers', but prior to execution of 'meshcat.servers.zmqserver'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
You can open the visualizer by visiting the following URL:
http://127.0.0.1:7000/static/
/usr/local/Cellar/pinocchio/2.5.2/lib/python3.9/site-packages/pinocchio/visualize/meshcat_visualizer.py:147: UserWarning: Error while loading geometry object: base_link_0
Error message:
No to_python (by-value) converter found for C++ type: boost::shared_ptr<hpp::fcl::CollisionGeometry>
  self.loadViewerGeometryObject(visual,pin.GeometryType.VISUAL,color)
```

This is on macOS where I installed pinocchio using homebrew.

Any idea how this could be fixed?

[1]: https://github.com/open-dynamic-robot-initiative/robot_properties_solo/blob/master/demos/demo_display_solo12_meshcat.py

jcarpent, 2021-01-13:
You did not install Pinocchio with the hpp-fcl support ;)

jcarpent, 2021-01-13:
OK. If it is related to homebrew, I will have a look.

jcarpent, 2021-01-13:
In between, you can also rely on `conda`.

jviereck, 2021-01-13:
Thanks a lot, the pointer to hpp-fcl helped!

It looks like hpp-fcl was installed but not linked. So after running

```bash
jviereck@rams:demos$ brew install hpp-fcl
Warning: gepetto/gepetto/hpp-fcl 1.6.0 is already installed, it's just not linked
You can use `brew link hpp-fcl` to link this version.
jviereck@rams:demos$ brew link hpp-fcl
```

things are working fine now for me. Thanks a lot!

closed 2021-01-13 by jviereck