stack-of-tasks/pinocchio #1372: num_max_contacts in collision requests

Diiso, 2021-01-19:
Hello,

I am trying to do contact related stuff with Pinocchio. For that, I need to compute collisions between shapes, and get information about their contact points. So far I have only been experimenting with a simple shpere and a plane to act as a floor, so I only needed one contact point, but I have switched to box and plane. The issue is the following: I can't seem to impose more than 1 contact point between two shapes. 
Here is an example which outputs 1 contact point when I would like to have more than 1 (which should be possible since these two boxes are interpenetrating):

```
import pinocchio as pin

# Model
model = pin.Model()

freeflyer = pin.JointModelFreeFlyer()
M = pin.SE3.Identity()
joint_id = model.addJoint(0, freeflyer, M, "joint_box") 
model.appendBodyToJoint(joint_id, pin.Inertia.FromEllipsoid(1, 0.1, 0.1, 0.1), M) 
data = model.createData()

# Geometry
geom_model = pin.GeometryModel()
box_shape = hppfcl.Box(0.4, 0.4, 0.4)
floor_shape = hppfcl.Box(3., 6., 0.1)

geom_floor = pin.GeometryObject("floor", 0, 0, floor_shape, M)
geom_floor.meshColor = np.array([0.2, 0.2, 1., 1.])
geom_model.addGeometryObject(geom_floor)

geom_box = pin.GeometryObject("box", 0, joint_id, box_shape, M) 
geom_box.meshColor = np.array([1.0, 0.2, 0.2, 1.]) 
geom_model.addGeometryObject(geom_box)

# Collisions
col_pair = pin.CollisionPair(0, 1)
geom_model.addCollisionPair(col_pair)

geom_data = geom_model.createData()
geom_data.collisionRequests[0].enable_contact = True # Enable the calculation of infos related to contact points (normal, position...)
geom_data.collisionRequests[0].num_max_contacts = 10

# Collision test
q = pin.utils.zero(model.nq)
pin.framesForwardKinematics(model, data, q)
pin.updateGeometryPlacements(model, data, geom_model, geom_data, q)
pin.computeCollisions(model, data, geom_model, geom_data, q, False)

cr = geom_data.collisionResults[0]
print("Num contact points: ", cr.numContacts())
```

Is there something I am doing wrong?
Thanks.

jmirabel, 2021-01-19:
You only get more contact points with BVH objects.

Primitive shapes are all convex volume (not only the surface of the shape). The intersection between two primitives is a convex volume. We could obviously return several contact points within intersection but we would need a strategy for finding them. At the moment, there are none.

Diiso, 2021-01-19:
I see. If I had cube1, cube2 and floor, I could get a contact point for cube1/floor and cube1/cube2 but I couldn't have n contact points for cube1/floor for example, because of the lack of contact point sampling strategy in the intersection volume. Thanks a lot for explaining @jmirabel.

closed 2021-01-20 by jcarpent