stack-of-tasks/pinocchio #1374: ABA overwrites data.a

gabrielebndn, 2021-01-20:
In order to store auxiliary accelerations ABA employs `data.a`, meaning this field is modified by the algorithm and left in an inconsistent state when the algorithm has terminated (the final value corresponds to the joint acceleration due to articular velocity with zero acceleration, plus gravity)

It should probably be replaced by `data.a_gf`, as done for instance by `rnea`. Notice that `computeABADerivatives` correctly employs `data.a_gf`

Another minor potential issue is that at the end of the algorithm `data.f` is filled with values which are not consistent with the final joint acceleration, which is not taken into account

jcarpent, 2021-01-20:
Some of your points have been handled in Pinocchio 3.x, but we won't release it because of lack of time.
This new release will mainly provide faster ABA and derivatives.

Concerning consistent output for forces, we've followed the original ABA algorithm, we do not provide joint forces. From the current state, it is VERY EASY to do it on YOUR SIDE. So, I won't consider this as an issue as you wrote, but rather as a feature!

gabrielebndn, 2021-01-20:
> From the current state, it is VERY EASY to do it on YOUR SIDE. So, I won't consider this as an issue as you wrote, but rather as a feature!

Yes, absolutely, I just opened the issue to let you know about it, but I can certainly fix it myself as soon as I have some time, I just wanted to let you know about it beforehand and I was mostly interested in what you thought about the forces, and I got my answer.

So, would you accept a PR simply replacing `data.a` with `data.a_gf`?

jcarpent, 2021-01-20:
Yes, sure.

PR #1376 (2021-01-21): [aba] Use data.a_gf instead of data.a
Fix #1374
closed 2021-01-21 by jcarpent