stack-of-tasks/pinocchio #1375: Fail to Run Meshcat Example

junhyeokahn, 2021-01-20:
Hello,
Thanks for maintaining this nice repo.
I just installed the package using conda, and trying to run some examples.
Especially, I am running two examples ```meshcat-viewer-dae.py``` and ```python meshcat-viewer.py```, but getting the warning below and the empty scene on the web. 
```
/opt/anaconda3/lib/python3.8/runpy.py:127: RuntimeWarning: 'meshcat.servers.zmqserver' found in sys.modules after import of package 'meshcat.servers', but prior to execution of 'meshcat.servers.zmqserver'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
You can open the visualizer by visiting the following URL:
http://127.0.0.1:7000/static/
```
Am I missing any procedure to run the example?

jcarpent, 2021-01-20:
How did you install meshcat?

junhyeokahn, 2021-01-20:
Yes I installed it by following ```pip install meschat```.

jcarpent, 2021-01-20:
You need to run it in an interactive manner, with:
```
python meshcat-viewer.py
```

junhyeokahn, 2021-01-20:
It works!
I really appreciate your help.

closed 2021-01-20 by junhyeokahn